marcelstoer/docker-nodemcu-build #1: Resulting image not starting correctly

Ampakinetic, 2015-10-31:
Hi Marcel,

Thank you for making this excellent tool.  I have got it to run and complile the binary files without any visible issues on my x64 Win7 setup, however after burning the integer version of dev140 to a 4Mb flash NodeMcu PCB I get the following out of the serial port but Lua doesn't start up at 9600 baud.  I've used your build tool to make dev140 on the same board without any trouble yesterday.

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

                               
                                                                                
load 0x40100000, len 28296, room 16 
                                          
tail 8
                                                                        
chksum 0xfd
                                                                   
load 0x00010003, len 131075, room 0 
                                          
                                              
```

marcelstoer, 2015-10-31:
I'm afraid I won't be of much help here as I don't have a Windows system for testing. Besides, since I have no idea what you changed in the firmware code you cloned from GitHub before you triggered the build I'd need a crystal ball to figure out what the root cause is.
However, in order to verify things are setup properly you could do the following:
- `git clone https://github.com/nodemcu/nodemcu-firmware.git`
- `cd nodemcu-firmware`
- `git checkout dev`
- edit the modules list in `app/include/user_modules.h`, leave all other files untouched.
- if you run `git status` now it should list `user_modules.h` as the only modified file (`modified:   app/include/user_modules.h`)
- run the Docker build and take note of the size of `bin/nodemcu_integer_dev.bin`
- now run a cloud build with exactly the same modules against the `dev` branch, the integer firmware binary should be the same size as `nodemcu_integer_dev.bin`

I don't know how familiar you are with the technologies involved here so forgive if this is all very obvious.

Ampakinetic, 2015-11-01:
Thank you very much for your response, I did exactly as you suggested, and to my surprise found that the cloud build (which is about 300 bytes different, probably because of the slightly larger welcome message) wouldn't run either.  

So the problem ended up being that esptool.py (which I normally don't use, but was using your suggested command line) was flashing the blue status light and returning success, but was burning both firmware versions incorrectly somehow.  

I could get both my custom build and the cloud build to run properly as soon as I used the NodeMcu .exe - so yay, I'm very happy I can do these custom builds now as I want to try adding support for the RC522 RFID module to Lua.

closed 2015-11-01 by Ampakinetic