marcelstoer/docker-nodemcu-build #2: Consistent output image names

pjsg, 2016-01-13:
It would be great if there was a link to the output image that was the same for each build (it can have the branch name). This would make the build/burn cycle easier as I don't need to change the input filename for the burner after each build. I was thinking that add a 'ln -f <name with time> <name without time>' would suffice.

marcelstoer, 2016-01-13:
@alkorin was requesting something similar the other day.

He proposed to support passing the image name to the Docker command like so `docker run -e "IMAGE_NAME=foo" ...`. I'll implement that (and a few more improvements) in the next days.

pjsg, 2016-01-14:
Also, can you save the mapfile along with the image so that I can find the right mapfile when I get a crash (now that the node.bootreason() returns pointers!!)

marcelstoer, 2016-01-14:
@pjsg not sure I understand. The `mapfile` is created in `app/`in your firmware directory. Maybe you were expecting it in `bin/` (like I did)?

pjsg, 2016-01-14:
Yes -- the mapfile is created in app/ but then promptly overwritten by the next build. You don't ever get to see the float version of the mapfile as it is overwritten by the integer build. What I would like is to have the mapfile copied into the bin directory with a name just like the bin file (but ending with .map or similar)

marcelstoer, 2016-01-14:
Ah, got it. Thanks for the clarification.

closed 2016-01-23 by marcelstoer

marcelstoer, 2016-01-23:
Due to Docker's odd interpretation of what `latest` [means](http://container-solutions.com/docker-latest-confusion/) you need to run `docker pull marcelstoer/nodemcu-build` explicitly to work with the updated image if Docker downloaded it previously.

commit 370cbefb (2016-01-23): Add 3 parameters and rescue mapfile before it's overwritten

fixes #2