marcelstoer/docker-nodemcu-build #4: docker buid error

vnmone, 2016-02-12:
on windows10
I act according to the instructions? but get next error
clone-repo(https://github.com/nodemcu/nodemcu-firmware) - > docker console - > cd to local coy - > 
docker run --rm -ti -v `pwd`:/opt/nodemcu-firmware marcelstoer/nodemcu-build

fatal: Not a git repository (or any parent up to mount point /opt/nodemcu-firmware)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).

marcelstoer, 2016-02-12:
I don't have a Win 10 installation to test so I won't be able to help you much. It's not a bug of the image but has something to do with Docker and the OS.

I get the exact same error on OS X if I clone the Git repository into the `/tmp` folder. So, I suspect there are certain folder/OS combinations which just don't work well with Docker.

closed 2016-02-19 by marcelstoer

marcelstoer, 2016-03-14:
There's a potential fix described in #7. As mentioned though, I can't verify myself.

MustafaFayez, 2016-08-23:
Hello, this error encountered to me, all I did is that I changed the dir of the repo from D:/ drive to C:/Users/<user>/ and it worked like a charm! of course I used the double slash rule:
docker run --rm -it -v //c/Users/<user>/nodemcu-firmware://opt/nodemcu-firmware marcelstoer/nodemcu-build
note: I use windows 7 64bit

slowWriting, 2016-10-14:
For anyone having a problem, you need to configure shared drives for Docker.
https://docs.docker.com/docker-for-windows/#shared-drives

pcgeek86, 2018-08-07:
The documentation for the Docker CLI on Windows is incorrect. Once I configured the "shared drive" in Docker for Windows, per @slowWriting directions, I used the following command line to build successfully:

```
docker run --rm -it -v c:/nodemcu:/opt/nodemcu-firmware marcelstoer/nodemcu-build
```

Also note that I cloned the Git repository into `c:/nodemcu` instead of my user directory.