marcelstoer/docker-nodemcu-build #5: Please combine the built spiffs image into the final .bin file

pjsg, 2016-02-18:
There is a commit about to be merged into nodemcu-firmware:dev that automates the building of the spiffs image containing files. Could you please add this file into the single .bin file that your build script generates?

Thanks

TerryE, 2016-02-19:
Should this be optional or mandatory?

marcelstoer, 2016-02-19:
I'll add it to https://github.com/marcelstoer/docker-nodemcu-build#options I guess.

marcelstoer, 2016-04-03:
Will be implemented when https://github.com/nodemcu/nodemcu-firmware/pull/1064 and/or https://github.com/nodemcu/nodemcu-firmware/issues/1164 are merged.

Just to be clear, the aim is to use [`srec_cat` in Dockerfile](https://github.com/marcelstoer/docker-nodemcu-build/blob/master/Dockerfile#L39) to combine and pad the `.bin`s so that the SPIFFS image ends up at 0x78000, right? I keep learning new stuff around you guys all the time :wink:.

pjsg, 2016-04-03:
My (not yet a PR)  nodemcu/nodemcu-firmware#1164 generates either one or two spiffs image files. One is to fit in a small flash (512k) and the other into a large flash (4M). The problem is that you don't want to flash the small one into a large flash as that limits the amount of free disk space. Flashing the large one takes a long time.  I've switched over to the Cesanta flasher: [flasher](https://github.com/cesanta/fnc/tree/master/common/platforms/esp8266/stubs) -- you just want the modified esptool.py. This goes significantly faster -- it takes me around 9 seconds to flash a 512k image.

Figuring out what people actually want with regard to this is a challenge.

mk-pmb, 2019-12-31:
Any progress on this? Is it still relevant?