marcelstoer/docker-nodemcu-build #7: Runnin the docker command on Windows

Kylir, 2016-03-02:
Hi there!

Thanks for this super useful container!
I managed to build my custom firmware and add support for a non conventional (128x32) oled screen.

I struggled a little bit to run the docker command on Windows so I thought I would share the simple trick I found. You need to specify the full path in the command and you need to add a double forward back slashes (`//`) in front of the two paths. The command is then:

````docker run --rm -it -v //c/Users/<user>/nodemcu-firmware://opt/nodemcu-firmware marcelstoer/nodemcu-build````

I can submit a pull request with an updated README file if it helps.

Thanks!

marcelstoer, 2016-03-02:
Jason, thanks a bunch for this! Just the other day I got a support request via email from a guy who struggled with the exact same problem. As I don't use Windows myself my hands were tied and I couldn't help, very frustrating. I'll update the README accordingly.

marcelstoer, 2016-03-02:
> double forward back slashes

Uuhhm, what's that? Wouldn't a forward back slash negate itself and thus become a pipe (`|`) :wink: ?

closed 2016-03-02 by marcelstoer

Kylir, 2016-03-02:
Hehehe :)
Silly me! Double forward slashes, of course!
It's horrible enough...

marcelstoer, 2016-03-16:
Jason, are you sure you need the `//` also in front of the "Docker path" i.e. the mount target? I think you only need them for the Windows path. It should be `//c/Users/<user>/nodemcu-firmware:/opt/nodemcu-firmware` no?

Kylir, 2016-03-21:
Hi there,
Sorry for the delay.
I just tested and I can use both: `/opt...` or `//opt...`. It does not make any difference for me.

marcelstoer, 2016-03-21:
Thanks for the confirmation! I updated README.md to document all the findings.

Sneftel, 2016-07-01:
An addendum to this: I had to give Docker access to the drive containing my repository, in Settings/Shared Drives, before the "Not a git repository" error would go away.

denalta, 2016-08-10:
First of all you need to give Docker access to the drive containing my repository, in Settings/Shared Drives!!! This helped me and then try to use the command line without --rm

docker run -it -v //c/users/<user>/nodemcu-firmware://opt/nodemcu-firmware marcelstoer/nodemcu-build

It checked on Windows 10!

Have a nice day.

commit 86645332 (2016-03-02): Add note for Windows users
fixes #7