marcelstoer/docker-nodemcu-build #8: Docker mounting volume on windows problem solved, but time sync issue?

robonut, 2016-03-15:
Hi all, I too have been struggling to get the docker command to run under windows. Issue #7 had the answer in using c:\users\<user>\nodemcu-firmware path on the host os (with //c/Users/<user>/nodemcu-firmware in the docker command), this works whereas other paths on the host os do not. I think it is something to do with automatic shares for c:\Users created by virtualbox guest additions, but not sure exactly why. 

After figuring that out, I could run the build, however it always fails at some point through with a similar output to;

xtensa-lx106-elf-ar ru .output/eagle/debug/lib/liblibc.a .output/eagle/debug/obj/c_ctype.o .output/eagle/debug/obj/c_math.o .output/eagle/debug/obj/c_stdio.o .output/eagle/debug/obj/c_stdlib.o .output/eagle/debug/obj/c_string.o
xtensa-lx106-elf-ar: creating .output/eagle/debug/lib/liblibc.a
xtensa-lx106-elf-ar: unable to rename '.output/eagle/debug/lib/liblibc.a'; reason: Operation not permitted
make[2]: *** [.output/eagle/debug/lib/liblibc.a] Error 1
make[2]: Leaving directory `/opt/nodemcu-firmware/app/libc'
make[1]: *** [.subdirs] Error 2
make[1]: Leaving directory `/opt/nodemcu-firmware/app'
make: *** [.subdirs] Error 2

It's not always the same file that it stops on (sometimes it is cjson.a or whatever), but it always is the renaming that drops the ball.

I think it is something to do with the virtual folder mount, as I also sometimes get time sync warnings like such and such file has timestamp 0.063s in the future. Anyone able to help here?

marcelstoer, 2016-03-21:
I'd surprise me if it was related to the time sync warning you mentioned. The error message "Operation not permitted" is actually pretty clear.

If you run the Docker command as usual but append " /bin/bash" you get a prompt *inside* the container that Docker created and started (based on the image). It allows you to poke around and run a few tests e.g. doing file system operations such as creating, removing and renaming files. That might give you a few more hints.

robonut, 2016-03-21:
Thanks Marcel.
I did actually get it going - but I have to admit I have no idea why the steps I took worked, they don't make a lot of sense.

What I had to do was copy the nodemcu firmware from the host os filesystem into the default docker vm filesystem (into the opt/nodemcu-firmware location). I still needed the -v option on the docker command, and still needed the firmware source in the host win7 filesystem under c:\Users etc. and it is in the host fs that the resultant bin files were created (as per normal). As I said, I have no idea why doing this had any effect - if anything I thought it would make  the mount part fail, but nevertheless this is what I did, and this is what let the build continue to the end and successfully produce output files.

closed 2016-03-22 by marcelstoer