marcelstoer/docker-nodemcu-build #11: wps support

apwiede, 2016-05-11:
For a project I need wps support in the nodemcu firmware. To enable that I have added wifi.wps.* commands in app/modules/wifi.c and I have added -lwps to the Makefile in modules. All builds well, but after flashing the firmware I get problems in a reboot every some seconds. Commenting out the calls to libwps.a and only having the lua commands in place makes the problem disappear. Is there a know issue why there is no wps support in nodemcu?

Cheers,
Arnulf 
(Arnulf Wiedemann)

apwiede, 2016-05-11:
forgot to mention: I have a repository clone and I am using the docker environment for building my firmware.
Also it is not the Makefile in modules but in app directory, which I have modified

Arnulf

marcelstoer, 2016-05-11:
That's an issue for the firmware repository. Keep http://nodemcu.readthedocs.io/en/dev/en/support/ in mind, though.

closed 2016-05-11 by marcelstoer