marcelstoer/docker-nodemcu-build #13: 'marcelstoer/nodemcu-build' shows a build error

mikeller, 2016-05-22:
```Build failed: The command '/bin/sh -c sudo apt-get update -y && sudo apt-get install -y wget unzip git make python-serial srecord bc' returned a non-zero code: 127```

Not really an issue with the code in this repo (as far as I can tell), but I couldn't find another way of making you aware of this.

marcelstoer, 2016-05-23:
> Not really an issue with the code in this repo

Correct, it's not. `RUN sudo apt-get ...` from the Dockerfile is turned into `/bin/sh -c sudo apt-get ...` by Docker.

closed 2016-05-23 by marcelstoer

mikeller, 2016-05-23:
Is there anything you can do to make the updated docker image available, or is it just a matter of waiting until docker fixes this?

marcelstoer, 2016-05-23:
> updated docker image

Which updated image? I didn't change anything.

mikeller, 2016-05-23:
The one where the BRANCH and BUILD_DATE variables are only set if they are needed, which came out of my pull request. At least the Dockerfile that is displayed on dockerhub looks like it's got there change.

marcelstoer, 2016-05-23:
Ah, I see...The PR was merged nearly two weeks ago and Docker Hub updated automatically. However, note that your local Docker installation won't automatically update an image it already downloaded (i.e. it won't check for an update from Docker Hub). You need to run `docker pull marcelstoer/nodemcu-build` manually to get the "latest" in that case.

mikeller, 2016-05-23:
Nope, doesn't seem to work. I did the `docker pull marcelstoer/nodemcu-build` (and tried again just now). I am still getting the `fatal: Not a git repository` error when trying to build with `marcelstoer/nodemcu-build`, so I assume Docker Hub serves the old image while it detects the updated one as 'broken'.