marcelstoer/docker-nodemcu-build #20: Odd error on W10

irjudson, 2016-08-09:
I get this in powershell, cmd prompt, and otherwise:

PS C:\Users\ivanjud> docker run --rm -it -v //c/Dev/nodemcu-firmware marcelstoer/nodemcu-build
fatal: Not a git repository (or any of the parent directories): .git
cp: cannot stat 'tools/esp-open-sdk.tar.gz': No such file or directory

Did I do something wrong?

PS -- Is there an IRC or Gitter channel for this?

marcelstoer, 2016-08-09:
> Did I do something wrong?

Looks like 😉 

> Docker runs on a VirtualBox VM which by default only shares the user directory from the underlying guest OS. On Windows that is `c:/Users/<user>` and on Mac it's `/Users/<user>`. Hence, you need to clone the  [NodeMCU firmware](https://github.com/nodemcu/nodemcu-firmware) repository to your *user directory*.

https://github.com/marcelstoer/docker-nodemcu-build#clone-the-nodemcu-firmware-repository

> PS -- Is there an IRC or Gitter channel for this?

No.

closed 2016-08-09 by marcelstoer

irjudson, 2016-08-09:
Have you tried this with the latest docker/windows? I still get the same problem when it's in C:\Users\ivanjud\Projects\nodemcu-firmware...
I'm using Docker 1.12.0-stable build 5968, 9cd1dc2

marcelstoer, 2016-08-09:
I'm on Mac and don't have access to post-Windows-7 machines to verify.

irjudson, 2016-08-09:
I'm on both and trying to make sure windows works! Thanks, I'll keep
poking...

--Ivan

Ivan R. Judson
irjudson.org
(425) 679-1375

marcelstoer, 2016-08-10:
Does https://github.com/marcelstoer/docker-nodemcu-build/issues/7#issuecomment-230044707 help?