marcelstoer/docker-nodemcu-build #22: spiffsimg build doesn't appear to work

mikep01, 2016-09-01:
Adding files to the ../local/fs directory doesn't seem to do anything, the file system remains empty.  Should the file section already be included in the bin output?

If I burn the bin files (00000,10000 individually the ESP goes into a loop with a bad file system.

Burning the file system bin to the indicated location seems to push the file system to a new address.

marcelstoer, 2016-09-01:
> Adding files to the ../local/fs directory doesn't seem to do anything

I don't expect this to work because it's what #5 requests.

closed 2016-09-01 by marcelstoer

mikep01, 2016-09-02:
Thanks for the reference!  I realized to late this is a build thing so nothing to do with this docker.