marcelstoer/docker-nodemcu-build #23: feature: option for build target

mikep01, 2016-09-01:
"clean" appears to be the default.  "all" and "flash" would be handy.

marcelstoer, 2016-09-01:
Something for the future...

Note that the default flash target was replaced in https://github.com/nodemcu/nodemcu-firmware/pull/1442. Eventually it'll be dropped altogether as we (i.e. NodeMCU team) don't want to host our own copy of `esptool.py`.

sanyaade-iot, 2018-05-17:
@marcelstoer : Is it possible also to have option for getting the source of the firmware being built for that user. That is, if I have selected certain options to be built for me. Then I can also get the source code of those firmware. This could be of a benefit to both developer and newbie who wants to study the code or continue to further the development locally.

I tried the on nodemcu-build, got the bin files via my email but no source which I was also expecting.
I know that the online system target non programmer/developer but it could be a very use tool for developers to a get a boiler template quickly and further the development locally. 

Also it be a tool for studying the firmware step by step through selected options/features when source code of those option can be obtained. Many thanks for this tool.

Hear from you soon!

God blesses!!!

Best regards,
Sanyaade

marcelstoer, 2018-07-01:
Wouldn't go well with the general direction this thing is moving -> closing

@sanyaade-iot you need the source as a basis for this Docker image to do anything, no source -> no binary. Hence, your request is implicitly fulfilled.

closed 2018-07-01 by marcelstoer