marcelstoer/docker-nodemcu-build #26: esp-open-sdk.tar.gz -> xz

CyanoFresh, 2017-01-25:
This commit breaks this script: https://github.com/nodemcu/nodemcu-firmware/commit/a3145da52bd5baa33906c68d7b83feefa2db1544
There is no file with gz format and I have this in console when run:
```
###\nodemcu-firmware>docker run -e "INTEGER_ONLY=1" --rm -it -v ###/nodemcu-firmware:/opt/nodemcu-firmware marcelstoer/nodemcu-build
cp: cannot stat 'tools/esp-open-sdk.tar.gz': No such file or directory
```

marcelstoer, 2017-01-25:
Was fixed a few days ago with a35ce20d566baea58db391b43b1ea496aae5b9a9

However, Docker doesn't automatically update your local image if there's an updated version on Docker Hub. So, you need to run `docker pull marcelstoer/nodemcu-build` manually once.

closed 2017-01-25 by marcelstoer

marcelstoer, 2017-01-26:
I just added a note to the README as you're likely not the only one running into this issue.

CyanoFresh, 2017-01-26:
ok, thanks. It helped