marcelstoer/docker-nodemcu-build #31: Suggestion: display selected modules in firmware startup for firmware generated with Docker image

lsr1, 2017-11-18:
Hi,

this is a suggestion to bring the docker building behavior near(er) to the cloud-building service (on https://nodemcu-build.com/). 

NodeMCU images built on the cloud service identify themselves on the ESP after startup as follows:

NodeMCU custom build by frightanic.com
	branch: master
	commit: 2e67ff5a639a13260fd4fb3c3b627ccdc2845616
	SSL: false
	modules: file,gpio,net,node,tmr,uart,wifi,ws2812,file,gpio,net,node,tmr,uart,wifi,file,gpio,net,node,tmr,uart,wifi
 build 	built on: 2017-09-10 12:53
 powered by Lua 5.1.4 on SDK 2.1.0(116b762)
lua: cannot open init.lua

Whereas NodeMCU images built with Docker identify themselves on the ESP after startup as follows:

NodeMCU 2.1.0 built with Docker provided by frightanic.com
	branch: master
	commit: 443e8219527f5b2190324a969a4586f9d3d731bf
	SSL: false
 build created on 2017-11-18 21:49
 powered by Lua 5.1.4 on SDK 2.1.0(116b762)
lua: cannot open init.lua

The difference is the display of the selected modules, which is IMHO incredible useful. I would submit a pull request, but I assume the code is already around somewhere (on the clould build service)?

marcelstoer, 2017-11-18:
> I assume the code is already around somewhere

It's not (I'd have reused it here 😉 ) as the cloud builder gets that list from the configuration the user created through the web UI.

lsr1, 2017-11-18:
Well s**t ;-)
Let's bring out sed then... 
Thank you for your reply!

PR #37 (2018-04-08): Add list of enabled modules to welcome message
Fixes #31
closed 2018-04-08 by marcelstoer