marcelstoer/docker-nodemcu-build #32: fatal: Not a git repository (or any parent up to mount point /opt/nodemcu-firmware)

woodcarving, 2018-01-01:
livecd Ubuntu 16.04 LTS, after installing git and docker:
```
cd ~
git clone https://github.com/nodemcu/nodemcu-firmware.git
```
```
Cloning into 'nodemcu-firmware'...
remote: Counting objects: 14202, done.
remote: Compressing objects: 100% (52/52), done.
remote: Total 14202 (delta 23), reused 30 (delta 12), pack-reused 14138
Receiving objects: 100% (14202/14202), 100.65 MiB | 3.67 MiB/s, done.
Resolving deltas: 100% (9499/9499), done.
Checking connectivity... done.
```
```
sudo docker run --rm -ti -v `pwd`:/opt/nodemcu-firmware marcelstoer/nodemcu-build
```
```
Unable to find image 'marcelstoer/nodemcu-build:latest' locally
latest: Pulling from marcelstoer/nodemcu-build
9fb6c798fa41: Pull complete 
3b61febd4aef: Pull complete 
9d99b9777eb0: Pull complete 
d010c8cf75d7: Pull complete 
7fac07fb303e: Pull complete 
a9b2cca3f218: Pull complete 
b3af545b8284: Pull complete 
d1dfec3e935f: Pull complete 
e8ef04736ed5: Pull complete 
bf66f7c97847: Pull complete 
Digest: sha256:763f25d549f9af009c0dfb1cfdb970f028dd90b1d94b71d39cd616ce82a73ca5
Status: Downloaded newer image for marcelstoer/nodemcu-build:latest
fatal: Not a git repository (or any parent up to mount point /opt/nodemcu-firmware)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
```
Ok..
```
cd ~
git init
cd /
sudo git init
```
Ok, one more time
```
cd ~
sudo docker run --rm -ti -v `pwd`:/opt/nodemcu-firmware marcelstoer/nodemcu-build
```
```
fatal: ambiguous argument 'HEAD': unknown revision or path not in the working tree.
Use '--' to separate paths from revisions, like this:
'git <command> [<revision>...] -- [<file>...]'
```
how to fix it and build? Thanks.

marcelstoer, 2018-01-01:
https://github.com/marcelstoer/docker-nodemcu-build#run-this-image-with-docker

> ...and **change to the NodeMCU firmware directory** (in the Docker console). Then run...

```
cd ~/nodemcu-firmware
docker...
```

closed 2018-01-01 by marcelstoer

rvalle, 2020-03-14:
I am having the same problem
it is not that I did not enter the firmware directory after the git clone, but rather:
```
[rafael@asusraf nodemcu-firmware]$ docker run --rm -ti -v $(pwd):/opt/nodemcu-firmware marcelstoer/nodemcu-build bash
root@f2e68fbc3563:/opt/nodemcu-firmware# ls -l
ls: cannot open directory '.': Permission denied
```

rvalle, 2020-03-14:
One would ask how is it that an image running as root cannot access any directory?
That rings the bell as SELINUX or similar

rvalle, 2020-03-14:
yes. 
```
sudo setenforce 0
```

I use fedora 31 that has SELINUX enabled, apparently.