marcelstoer/docker-nodemcu-build #49: fatal: Not a git repository

loadedtech, 2018-10-21:
Firstly, thanks for your great work ... 

I'm trying to do a build on OSX High Sierra using Docker Community Edition v18.06.1 and am getting the error when I try docker run --rm -ti -v 'pwd':/opt/nodemcu-firmware marcelstoer/nodemcu-build build:
fatal: Not a git repository (or any parent up to mount point /opt/nodemcu-firmware)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).

docker run --rm -ti -v 'pwd':/opt/nodemcu-firmware marcelstoer/nodemcu-build shows the "Welcome to the Docker image ... " message

I cloned the nodemcu-firmware repo (into my user folder), checked permissions, checked ownership, git status ... just can't figure it out. 

Does anyone have any clues, suggestions or things to check?

loadedtech, 2018-10-21:
Ok, so I made progress by replacing 'pwd' with the full path to my nodemcu-firmware directory ... now all good.

I don't know why 'pwd' didn't work

marcelstoer, 2018-10-21:
> I don't know why 'pwd' didn't work

Because you need to use back ticks rather than single quotes around `pwd`, look at https://github.com/marcelstoer/docker-nodemcu-build#run-this-image-with-docker-to-create-the-firmware carefully.

closed 2018-10-21 by marcelstoer

loadedtech, 2018-10-21:
FYI the backticks threw a docker error:
opt/lua  marcelstoer/nodemcu-build lfs-im
docker: invalid reference format: repository name must be lowercase.
See 'docker run --help'.

HHHartmann, 2018-10-22:
Do you have a space in your path? 
In that case you would need to add double quotes around the pwd.
Like ''"`pwd`"'

loadedtech, 2018-10-22:
yes, the path does have a space .. will try the double quotes

FYI, I posted a new issue / suggestion but for some reason it didn’t save / show up for me when I reloaded … did you get /see the issue?

Once again, thanks for your great work.

Hugh

instructions for "Flashing the built binary" with SPIFFS_FIXED_LOCATION
I think the instructions for "Flashing the built binary" might / should reference the setting:
#define SPIFFS_FIXED_LOCATION 0x100000

Presumably the section should include something like:
If you set your build to a fixed location using something like:
#define SPIFFS_FIXED_LOCATION 0x100000
then the command to flash the image will become
esptool.py --port <USB-port-with-ESP8266> write_flash 0x100000 <NodeMCU-firmware-directory>/bin/nodemcu_[integer|float]_<Git-branch>.bin

instructions for "Flashing the built binary" with SPIFFS_FIXED_LOCATION #51
 Open	loadedtech <https://github.com/loadedtech> opened this issue 5 hours ago · 0 comments
Comments
Assignees
No one assigned
Labels
None yet
Projects
None yet
Milestone
No milestone
Notifications
 Unsubscribe
You’re receiving notifications because you authored the thread.
1 participant
 <https://github.com/loadedtech>
 <https://github.com/loadedtech>
  
loadedtech <https://github.com/loadedtech> commented 5 hours ago <https://github.com/marcelstoer/docker-nodemcu-build/issues/51#issue-372373641>
I think the instructions for "Flashing the built binary" might / should reference the setting:
#define SPIFFS_FIXED_LOCATION 0x100000

Presumably the section should include something like:
If you set your build to a fixed location using something like:
#define SPIFFS_FIXED_LOCATION 0x100000
then the command to flash the image will become
esptool.py --port <USB-port-with-ESP8266> write_flash 0x100000 <NodeMCU-firmware-directory>/bin/nodemcu_[integer|float]_<Git-branch>.bin

    ----------------------------------------- 
      Hugh Morrow                    iconic 
                         +61 (0)411 477 487
                         hugh@iconic.com.au
    -----------------------------------------

marcelstoer, 2018-10-22:
> for some reason it didn’t save / show up for me when I reloaded

Yeah, GitHub was instable the whole day https://status.github.com/messages