marcelstoer/docker-nodemcu-build #54: Disabled LUA_NUMBER_INTEGRAL Ignored by LFS Image Build Step

jmd13391, 2018-11-10:
### Expected Behavior
lfs-image step should not attempt to build an integer image when app/include/user_config.h: #define LUA_NUMBER_INTEGRAL is commented-out.

### Actual Behavior
lfs-image step ignores #define LUA_NUMBER_INTEGRAL and attempts to build the integer image. The attempt fails and throws an error:

docker run --rm -ti -e "IMAGE_NAME=test" -v "{PWD}":/opt/nodemcu-firmware -v "{PWD}/user-apps":/opt/lua marcelstoer/nodemcu-build lfs-image

> Adding files: ../lua/_init.lua ../lua/_main.lua ../lua/_preload.lua
> creating LFS_float_test.img
> **creating LFS_integer_test.img**
> **./luac.cross.int: ../lua/_main.lua:1: malformed number near '27.60'**

Offending Source Code Line:
`local BFCVAL = 27.60`

### References & Additional Notes

REF: [https://github.com/marcelstoer/docker-nodemcu-build/](https://github.com/marcelstoer/docker-nodemcu-build/)

> INTEGER_ONLY and FLOAT_ONLY are not supported anymore. Please configure LUA_NUMBER_INTEGRAL in app/include/user_config.h as described above.

app/include/user_config.h:
`//#define LUA_NUMBER_INTEGRAL`

HHHartmann, 2018-11-10:
The idea was to build an image for whatever versions of luac.cross are abailable in the source code root directory.

So you seem to have luac.cross for float builds and luac.cross.int left over from an int build.

As your sources contain floating point numbers which are not supported in int builds it fails.

I felt it was a good idea to build all available types.

jmd13391, 2018-11-10:
> _you seem to have luac.cross for float builds and luac.cross.int left over from an int build._

Yup... that was it. A complete miss on my part. I removed the offending luac.cross.int file and I am no longer receiving the error.

I can't help but think that we could use some sort of housecleaning on the prior "Build the Firmware" step to remove old luac.cross[.int] files (if they exist) prior to starting a build.

HHHartmann, 2018-11-10:
Maybe you are right. At least it might be cleaned on ```make clean``` But that's rather to be discussed over at nodemcu-firmware.

closed 2018-11-13 by jmd13391