marcelstoer/docker-nodemcu-build #58: sed error

raghur, 2018-12-05:
I'm on linux. I get the following error with docker when I try to build after configure.

```
xxxx@home-desktop  ~/code/esp32/tools/nodemcu-firmware   dev-esp32 ●  docker run --rm -ti -v `pwd`:/opt/nodemcu-firmware marcelstoer/nodemcu-build build
sed: -e expression #1, char 99: unknown option to `s'
```

marcelstoer, 2018-12-05:
What are the special characters in the command you posted?

Could you try `docker run --rm -ti -v `pwd`:/opt/nodemcu-firmware marcelstoer/nodemcu-build /bin/bash` and then run the `sed` commands from https://github.com/marcelstoer/docker-nodemcu-build/blob/master/build-esp32 manually to help me find out _which_ command actually fails?

raghur, 2018-12-05:
> What are the special characters in the command you posted?
that's just my zsh prompt. command being run is everything after docker.

here's the sed commands

```
root@fc49106ab9f8:/opt/nodemcu-firmware# BRANCH="$(git rev-parse --abbrev-ref HEAD | sed -r 's/[\/\\]+/_/g')"
root@fc49106ab9f8:/opt/nodemcu-firmware# echo $BRANCH
dev-esp32
root@fc49106ab9f8:/opt/nodemcu-firmware# SDK_VERSION="$(git submodule status|grep esp32|sed -r 's/.*\((.+)\)/\1/')"
root@fc49106ab9f8:/opt/nodemcu-firmware# echo $SDK_VERSION
-fc19954376654c3e1b4bec5c25cfa5ca241d39d1 sdk/esp32-esp-idf
root@fc49106ab9f8:/opt/nodemcu-firmware# sed -i '/#define LUA_RELEASE[[:space:]]/ s/$/ " on ESP-IDF '"$SDK_VERSION"'"/g' "$lua_header_file"
sed: -e expression #1, char 99: unknown option to `s'
```

raghur, 2018-12-05:
I haven't made any changes at all... 

Meanwhile, I did edit the build-esp32 file and removed the line so I could move forward. I'm puzzled as well!

I'll check lua.h again and just to make sure, run a fit clean as well tomorrow and report back

raghur, 2018-12-06:
No edits in the working directory other than ` modified:   components/platform/include/user_version.h`

imediabank, 2018-12-14:
i am a newbie and have same problem sed: -e expression #1, char 99: unknown option to `s'

can anyone show my detail how to fixed it ?

hk-badbuta, 2018-12-18:
I have checked the file: /opt/build-esp32.  I find that the sed line contained $SDK_VERSION causes the problem.  it is because the string $SDK_VERSION contains slash '/' character that conflict with the search-and-replace syntax.   I try to replace the '/' to '|' and it works.  Here is the fixed line I use:

# append " on ESP-IDF xxx" to the Lua version
# sed -i '/#define LUA_RELEASE[[:space:]]/ s/$/ " on ESP-IDF '"$SDK_VERSION"'"/g' "$lua_header_file"
sed -i '/#define LUA_RELEASE[[:space:]]/ s|$| " on ESP-IDF '"$SDK_VERSION"'"|g' "$lua_header_file"

marcelstoer, 2018-12-18:
Thanks for the proposed fix but `$SDK_VERSION` is not supposed to contain a '/'. Can you insert a `echo "$SDK_VERSION"` to show what its value is?

marcelstoer, 2018-12-19:
@imediabank this is unrelated to the issue we're discussing here pls create a new ticket

-> I will delete the comment

However, I have the feeling yours is completely unrelated to this Docker image.

hk-badbuta, 2018-12-19:
the value of "$SDK_VERSION" is ```sdk/esp32-esp-idf```

How I get this value:
``` bash
$ cd {path to nodemcu-firmware}
$ docker run --rm -ti -v `pwd`:/opt/nodemcu-firmware marcelstoer/nodemcu-build bash
root@xxxx: vi /opt/build-esp32
...
root@xxxx: build
```

The version of nodemcu-firmware I use: 
```
On branch dev-esp32
Your branch is up to date with 'origin/dev-esp32'.
```

My OS is Ubuntu 18.04.1 LTS

marcelstoer, 2018-12-20:
Ok, found it...it's a user error but a mean one. The hint was actually right there in @raghur's 2nd comment; I just didn't see it.

```
root@fc49106ab9f8:/opt/nodemcu-firmware# SDK_VERSION="$(git submodule status|grep esp32|sed -r 's/.*\((.+)\)/\1/')"
root@fc49106ab9f8:/opt/nodemcu-firmware# echo $SDK_VERSION
-fc19954376654c3e1b4bec5c25cfa5ca241d39d1 sdk/esp32-esp-idf
``` 

`$SDK_VERSION` obviously has the wrong value, something which @hk-badbuta also reported. It should be something like `v3.3-dev-149-g1c4d9f2` (what I have here). This happens if you don't tell Git to resolve submodules upon checkout/clone as per https://nodemcu.readthedocs.io/en/dev-esp32/en/build/#linux-build-environment

I will add a note to https://github.com/marcelstoer/docker-nodemcu-build#build-for-esp32

closed 2018-12-20 by marcelstoer

raghur, 2018-12-21:
Ok - So I think there's more to this. I checked my repo... Interestingly, from my host if I run `git submodule status` I get
```
1c4d9f2acacc666a4eb02f1450abd400e7df58c3 sdk/esp32-esp-idf (v3.3-dev-149-g1c4d9f2ac)

# git version
raghu@home-desktop  ~/code/esp32/tools/nodemcu-firmware   dev-esp32 ●  git --version       
Found existing alias for "git". You should use: "g"
git version 2.17.1
```

However, when I run the same thing inside the docker container, I get
```
root@14e279a1a318:/opt/nodemcu-firmware# git submodule status | grep esp32
-1c4d9f2acacc666a4eb02f1450abd400e7df58c3 sdk/esp32-esp-idf

root@14e279a1a318:/opt/nodemcu-firmware# git --version
git version 1.9.1
```

Within the container, if I run `git submodule update --init` somehow seems to fix things - after that `git submodule status` shows the tag version in brackets. Perhaps a good idea to include that in the build-esp32 script?

marcelstoer, 2018-12-24:
@raghur thanks for your detailed feedback. The behavior you're seeing is different from the one I saw when I briefly tested on Windows. So, _does_ seem to be related to #57 after all.

raghur, 2018-12-25:
I see (or probably I dont).. in my case, I'm on Ubuntu KDE neon 18.04, if that's of any use.

dacodekid, 2019-02-08:
@marcelstoer , I'm facing the same issue on MacOS (followed the exact procedure from README) and was able to fix it based @raghur 's finding.

```
docker run --rm -ti -v `pwd`:/opt/nodemcu-firmware marcelstoer/nodemcu-build bash
git submodule update --init
exit
```

and it works.

jcwren, 2019-03-16:
Same for me as @dacodekid, along with the suggested fix working.

coolwolf, 2019-05-18:
@dacodekid thank you very much. this solve my problem.