marcelstoer/docker-nodemcu-build #60: Allow selecting lua files which will be added to LFS image

HHHartmann, 2018-12-20:
I would like to add a feature to allow selecting only specific files to add to the LFS image.

There could be a file containing a list of files to add using unix wildcards if desired.
Could be useful for complete projects containing an init.lua and other files. Obviously the init.lua does not need to be in the LFS.

The file could look like this:
``` sh
startup.lua  main.lua
modules/*.lua  drivers/*.lua   misc/logging.lua
````

the filename would be passed as parameter to lfs-image.
If given use its content else pack all files.

HHHartmann, 2019-01-03:
OK I got a working version. But it only works for files containing no characters which would have to be escaped. Especially spaces.
Should I still prepare a PR?

marcelstoer, 2019-01-03:
I haven't given this much thought so far, sorry. Be as restrictive as you like. My approach would probably be
a) find a Linux tool that you can feed this config file and that returns a fully-qualified list of files
b) see if that config file with a defined name exists in a defined location e.g. `/lfs-files.something` 
  - don't give the user choices
  - `.something` (i.e. the file type) is defined by the outcome of a), could be JSON, YAML, etc.