marcelstoer/docker-nodemcu-build #66: what to do with the LFS image?

misterinterrupt, 2019-05-08:
Hi there, thanks for all your awesome work! I am just getting into working with the nodemcu 0.9 and I have been able to figure out what to do with the LFS image..
the README shows the use of the docker image to 
1. build the firmware, this works for me
2. flash the firmware, also works for me
3. build an LFS image, I can build it, but what am I supposed to do with it?

~at one point, I thought the firmware build would take care of the SPIFFS/LFS needs..
I have the default `init.lua` in the `local/lua` folder along with the `credentials.lua` file
I also put a copy of an mqtt lua example into the` local/fs` dir 

once I built and flashed the firmware, I run nodemcu-tool fsinfo  and see this:
```
$ nodemcu-tool --port /dev/ttyUSB0 --debug --io-debug fsinfo --raw --connection-delay=1500
[config]      ~ debug mode enabled 
[config]      ~ io-debug mode enabled - rx/tx messages are passed to logger 
[config]      ~ baudrate             = 115200 (type:string) 
[config]      ~ port                 = /dev/ttyUSB0 (type:string) 
[config]      ~ connectionDelay      = 1500 (type:string) 
[config]      ~ minify               = false (type:boolean) 
[config]      ~ compile              = false (type:boolean) 
[config]      ~ keeppath             = false (type:boolean) 
[config]      ~ remotename           = null (type:object) 
[config]      ~ run                  = false (type:boolean) 
[config]      ~ all                  = false (type:boolean) 
[config]      ~ json                 = false (type:boolean) 
[config]      ~ raw                  = true (type:boolean) 
[config]      ~ softreset            = false (type:boolean) 
[serialport]  ~ TX: print("echo1337")
[serialport]  ~ RX: > 
> 
> print("echo1337")
[serialport]  ~ RX: echo1337
[serialport]  ~ RX: > 
> 
>
[serialport]  ~ TX: print("echo1337")
[serialport]  ~ RX: > 
> 
> print("echo1337")
[serialport]  ~ RX: echo1337
[serialport]  ~ RX: > 
> 
>
[serialport]  ~ TX: print("echo1337")
[serialport]  ~ RX: print("echo1337")
[serialport]  ~ RX: echo1337
[serialport]  ~ RX: >
[serialport]  ~ TX: print(node.info());
[serialport]  ~ RX: print(node.info());
[serialport]  ~ RX: 2
[serialport]  ~ RX: 2
[serialport]  ~ RX: 0
[serialport]  ~ RX: 9171440
[serialport]  ~ RX: 1458400
[serialport]  ~ RX: 4096
[serialport]  ~ RX: 2
[serialport]  ~ RX: 40000000
[serialport]  ~ RX: >
[NodeMCU-Tool]~ Connected 
[device]      ~ Arch: esp8266 | Version: 2.2.0 | ChipID: 0x8bf1f0 | FlashID: 0x1640e0 
[serialport]  ~ TX: print(file.fsinfo())
[serialport]  ~ RX: print(file.fsinfo())
[serialport]  ~ RX: 3198242
[serialport]  ~ RX: 0
[serialport]  ~ RX: 3198242
[serialport]  ~ RX: >
[serialport]  ~ TX: local l = file.list();for k,v in pairs(l) do uart.write(0,k..":"..v..";") end print("")
[serialport]  ~ RX: local l
[serialport]  ~ RX: = file.list();for k,v in pairs(l) do uart.write(0,k..":"..v..";") end print("")
[serialport]  ~ RX: 
[serialport]  ~ RX: >
[NodeMCU-Tool]~ disconnecting 

```
no mention of the files or the free space, like I have seen in examples...

I need a high level explanation of what is supposed to be done with the nodemcu regarding the filesystem stuff..  there seems to be docs talking about all the parts, but what is the LFS for? do you need one on the nodemcu in order to run lua code on it? 

Thanks for any pointers!

HHHartmann, 2019-05-09:
What build does is 
- Generate an LFS image of all files in the lua folder and write it to the fs folder.
- Add all files in the fs folder to the SPIFFS image.

So you would need the init.lua in fs and the other files in the lua folder.
To "install" the LUA image you need to ```node.flashrestore("LFS.img")```. This will write the LFS image to the reserved space and reboot.
If it does not reboot an explanation is returned by node.flashrestore. 
On success the LFS.img file can then be deleted.

To see how to activate LFS you can have a look at ```nodemcu-firmware\lua_examples\lfs\_init.lua``` which gives usefull additions. It tweaks dofile, loadfile and require to load stuff from the LFS image. Put it in the LFS image too.
Also have a look at ```nodemcu-firmware\lua_examples\lfs\lfs_fragments.lua``` at the bottom to see how to tweak you init.lua to activate it all.

So the only file you need outside the LFS image is ```init.lua```.

marcelstoer, 2019-05-09:
@HHHartmann I'm grateful for you to pick up this support case, thank you. I will close it as doesn't concern this project but we can still continue here. 

@misterinterrupt are you familiar with https://nodemcu.readthedocs.io/en/latest/lfs/?

closed 2019-05-09 by marcelstoer

misterinterrupt, 2019-05-09:
thanks a ton for your help @HHHartmann ! looks like I was pretty close.
@marcelstoer apologies if this is the wrong place to ask.
I did read through that LFS doc page a couple times, but I didn't quite understand the 'why' of doing it or the necessity of it.. It seems like I do need to set up a filesystem. I just couldn't find a doc that communicated something very high level like the list of steps that I put above, at least not that included anything beyond build and flash the firmware.

marcelstoer, 2019-05-09:
Over in the firmware project developers (and we ourselves) noticed that there's quite a large gap between the "First steps" document and the LFS whitepaper. It is being addressed in https://github.com/nodemcu/nodemcu-firmware/issues/2656. Somewhere in the middle of that issue a user proposed some instructions which are tailored for Windows.

misterinterrupt, 2019-05-09:
hmm okay, if its alright, I can put together my notes and submit a linux-focused doc