marcelstoer/docker-nodemcu-build #70: ESP32 build problems on OS X

loadedtech, 2019-07-29:
I've encountered two problems building the firmware on OS X.

First, the documentation says to checkout the dev-esp32 branch and then do a "git submodule update --recursive".  But because the esp32-esp-idf submodule hasn't been initialised it doesn't update ... I first had to run "git submodule init"

Second, on trying to run configure-esp32 the first time I get a permissions error:
    2019-07-29 22:12:18 (91.2 KB/s) - '/opt/nodemcu-firmware/cache/toolchain-esp32-linux-x86_64-20181106.0.tar.xz' saved [28975720/28975720]
    Uncompressing toolchain
    tar: esp32-linux-x86_64-20181106.0/build.log.bz2: Cannot open: Permission denied
I think it first failed to decompress the toolchain archive and then when I did it manually I think I got the symbolic links (eg for xtensa-esp32-elf-ar) messed up.

marcelstoer, 2019-07-30:
> I first had to run "git submodule init"

The documentation also says to `git clone --recurse-submodules` (same chapter) which void avoid having to `git submodule init`.

> tar: esp32-linux-x86_64-20181106.0/build.log.bz2: Cannot open: Permission denied

Ah, interesting. We're battling with this in the xenial branch (upgrading Ubuntu from trusty): https://github.com/marcelstoer/docker-nodemcu-build/commit/a4087c64a3046625b828c746e138140974a08973
@HHHartmann this suggests that the behavior _we_ are seeing is unrelated to xenial vs trusty.

closed 2019-07-30 by marcelstoer

loadedtech, 2019-07-30:
> The documentation also says to git clone --recurse-submodules (same chapter) which void avoid having to git submodule init.
Sorry, I should have been clearer ... the docs say to clone, then checkout the dev-esp32 branch and then update ... but this order doesn't work.  You'd need to checkout the branch then clone

marcelstoer, 2019-08-01:
https://github.com/nodemcu/nodemcu-firmware/pull/2869 was just merged.