marcelstoer/docker-nodemcu-build #73: Support the new firmware info mechanism

marcelstoer, 2019-08-06:
Build support for https://github.com/nodemcu/nodemcu-firmware/pull/2830 into `build-esp8266`.

Set the environment variable `USER_PROLOG`.

@HHHartmann do you think it's feasible/acceptable to just remove 70% of the stuff in `build-esp8266` and thereby no longer altering the welcome message for the frozen 1.5.x branch? It would be a relief to get rid of all that code 😉

HHHartmann, 2019-08-07:
I would give it some time before removing the 70% thingy. Not everyone updates to the newest master all the time. Maybe one or two releases.

marcelstoer, 2019-08-07:
Hhhmm, alright...then indeed I'll need to create this one big if-else probing for `NODE_VERSION_LONG` or `tools/update_buildinfo.sh` as discussed before.

HHHartmann, 2019-08-07:
You could also rely on the`# define` trick for the time being. Setting USER_PROLOG won't hurt anybody.

HHHartmann, 2019-08-07:
But then you would need another docker image update

commit de49981b (2019-08-11): Support the new firmware build info mechanism

Fixes #73
closed 2019-08-11 by marcelstoer