marcelstoer/docker-nodemcu-build #77: build-esp32 fails with "project.mk: No such file or directory"

joysfera, 2019-09-13:
Decided to try working around the #76 and finally building an ESP32 image but I've just run into a different issue (this is on another computer, but everything's up-to-date) that ends with

  `project.mk: No such file or directory`

I'd rather include the whole bash session:

```
/tmp$ git clone --recurse-submodules https://github.com/nodemcu/nodemcu-firmware.git
Cloning into 'nodemcu-firmware'...
remote: Enumerating objects: 9, done.
remote: Counting objects: 100% (9/9), done.
remote: Compressing objects: 100% (9/9), done.
remote: Total 18985 (delta 0), reused 6 (delta 0), pack-reused 18976
Receiving objects: 100% (18985/18985), 107.29 MiB | 1.88 MiB/s, done.
Resolving deltas: 100% (12867/12867), done.
Checking connectivity... done.
Submodule 'app/u8g2lib/u8g2' (https://github.com/olikraus/U8g2_Arduino.git) registered for path 'app/u8g2lib/u8g2'
Submodule 'app/ucglib/ucg' (https://github.com/olikraus/Ucglib_Arduino.git) registered for path 'app/ucglib/ucg'
Cloning into 'app/u8g2lib/u8g2'...
remote: Enumerating objects: 109, done.
remote: Counting objects: 100% (109/109), done.
remote: Compressing objects: 100% (59/59), done.
remote: Total 8971 (delta 41), reused 58 (delta 7), pack-reused 8862
Receiving objects: 100% (8971/8971), 44.27 MiB | 3.23 MiB/s, done.
Resolving deltas: 100% (4869/4869), done.
Checking connectivity... done.
Submodule path 'app/u8g2lib/u8g2': checked out '2ee84c8f14adaa8fd1ebfe091c4de348c5474b18'
Cloning into 'app/ucglib/ucg'...
remote: Enumerating objects: 376, done.
remote: Total 376 (delta 0), reused 0 (delta 0), pack-reused 376
Receiving objects: 100% (376/376), 5.18 MiB | 1.77 MiB/s, done.
Resolving deltas: 100% (213/213), done.
Checking connectivity... done.
Submodule path 'app/ucglib/ucg': checked out 'e21641a6c1ddb0e71f7b9e01501fa739786c68b1'

/tmp$ cd nodemcu-firmware/

/tmp/nodemcu-firmware$ git checkout dev-esp32
warning: unable to rmdir app/u8g2lib/u8g2: Adresář není prázdný
warning: unable to rmdir app/ucglib/ucg: Adresář není prázdný
Branch dev-esp32 set up to track remote branch dev-esp32 from origin.
Switched to a new branch 'dev-esp32'

/tmp/nodemcu-firmware$ git submodule update --recursive

/tmp/nodemcu-firmware$ docker run --rm -ti -v `pwd`:/opt/nodemcu-firmware marcelstoer/nodemcu-build configure-esp32
wget --tries=10 --timeout=15 --waitretry=30 --read-timeout=20 --retry-connrefused https://github.com/jmattsson/esp-toolchains/releases/download/linux-x86_64-20181106.1/toolchain-esp32-linux-x86_64-20181106.1.tar.xz -O /opt/nodemcu-firmware/cache/toolchain-esp32-linux-x86_64-20181106.1.tar.xz || { rm -f "/opt/nodemcu-firmware/cache/toolchain-esp32-linux-x86_64-20181106.1.tar.xz"; exit 1; }
--2019-09-13 08:34:43--  https://github.com/jmattsson/esp-toolchains/releases/download/linux-x86_64-20181106.1/toolchain-esp32-linux-x86_64-20181106.1.tar.xz
Resolving github.com (github.com)... 140.82.118.4
Connecting to github.com (github.com)|140.82.118.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github-production-release-asset-2e65be.s3.amazonaws.com/156146883/53b62800-b479-11e9-8165-194de6d7e707?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20190913%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20190913T083443Z&X-Amz-Expires=300&X-Amz-Signature=74517e23f3897918c760fe8f89f8f6f56aa47dc74d4cf8ce574cad4f12eaad92&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Dtoolchain-esp32-linux-x86_64-20181106.1.tar.xz&response-content-type=application%2Foctet-stream [following]
--2019-09-13 08:34:44--  https://github-production-release-asset-2e65be.s3.amazonaws.com/156146883/53b62800-b479-11e9-8165-194de6d7e707?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20190913%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20190913T083443Z&X-Amz-Expires=300&X-Amz-Signature=74517e23f3897918c760fe8f89f8f6f56aa47dc74d4cf8ce574cad4f12eaad92&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Dtoolchain-esp32-linux-x86_64-20181106.1.tar.xz&response-content-type=application%2Foctet-stream
Resolving github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)... 52.216.146.67
Connecting to github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)|52.216.146.67|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 28337296 (27M) [application/octet-stream]
Saving to: '/opt/nodemcu-firmware/cache/toolchain-esp32-linux-x86_64-20181106.1.tar.xz'

/opt/nodemcu-firmware/cache/toolchain-esp32-linux-x86_64-20 100%[=========================================================================================================================================>]  27.02M  1.78MB/s    in 18s     

2019-09-13 08:35:02 (1.49 MB/s) - '/opt/nodemcu-firmware/cache/toolchain-esp32-linux-x86_64-20181106.1.tar.xz' saved [28337296/28337296]

Uncompressing toolchain
# the archive contains ro files
Setting IDF_PATH and re-invoking...
make[1]: Entering directory '/opt/nodemcu-firmware'
Makefile:42: /opt/nodemcu-firmware/sdk/esp32-esp-idf/make/project.mk: No such file or directory
make[1]: *** No rule to make target '/opt/nodemcu-firmware/sdk/esp32-esp-idf/make/project.mk'.  Stop.
make[1]: Leaving directory '/opt/nodemcu-firmware'
Makefile:18: recipe for target 'menuconfig' failed
make: *** [menuconfig] Error 2

```

joysfera, 2019-09-13:
BTW, I followed the `git pull` instructions from here:
https://hub.docker.com/r/marcelstoer/nodemcu-build/#clone-the-nodemcu-firmware-repository
and then the build fails as shown above.

But if I followed a different instructions for `git pull` from here:
https://nodemcu.readthedocs.io/en/dev-esp32/build/#setting-up-the-repository

then the initial make did not fail and I reached the menuconfig.

marcelstoer, 2019-11-16:
I just ran the documented sequence of commands (as per _this_ project's documentation). It works for me - I typed the wrong command but the scripts guided me to the right one 😃 

```
/tmp > git clone --recurse-submodules https://github.com/nodemcu/nodemcu-firmware.git
...
/tmp > cd nodemcu-firmware/
/tmp/nodemcu-firmware (master) > git checkout dev-esp32
...
/tmp/nodemcu-firmware (dev-esp32) > git submodule update --recursive
...
/tmp/nodemcu-firmware (dev-esp32) > docker run --rm -ti -v `pwd`:/opt/nodemcu-firmware marcelstoer/nodemcu-build build
ERROR: `sdkconfig` file missing or not accessible, run the `configure-esp32` command first
/tmp/nodemcu-firmware (dev-esp32) > docker run --rm -ti -v `pwd`:/opt/nodemcu-firmware marcelstoer/nodemcu-build configure-esp32

closed 2019-11-16 by marcelstoer