marcelstoer/docker-nodemcu-build #79: LFS Compiler Error

chathurangawijetunge, 2019-10-19:
When i do 
docker run --rm -ti -v `pwd`:/opt/nodemcu-firmware -v /home/pi/scw:/opt/lua marcelstoer/nodemcu-build lfs-image

I only get error 
Standard_init_linux.go:221: exec user process caused "exec format error"

HHHartmann, 2019-10-19:
Does your socket image work correctly? 
Try
```
docker run --rm -ti -v pwd:/opt/nodemcu-firmware -v /home/pi/scw:/opt/lua marcelstoer/nodemcu-build
```

chathurangawijetunge, 2019-10-19:
No getting the same error

HHHartmann, 2019-10-19:
That means your docker image is not correct.
I'm not sure about this but I guess that you need to build your own docker image as described in https://github.com/marcelstoer/docker-nodemcu-build/blob/master/Dockerfile
Don't worry, its just few easy steps to follow.
That might help or be a waist of bandwidth.
The reasoning behind this is that the raspberry pi has a different processor architecture than e.g. windows.

HHHartmann, 2019-10-19:
Ah you could also try starting one of the stock images (from the docker repository) to verify that you docker installation works correctly.

HHHartmann, 2019-10-19:
You will also have to download and build the firmware first to build the luac.cross binary as it is not included in the docker image. 
If you are using an integer firmware be sure to configure it in `user_config.h`. Uncoment `#define LUA_NUMBER_INTEGRAL`

marcelstoer, 2019-10-19:
> I'm using raspberry pi

That latest edit is essential as Gregor already mentioned (thanks!). I can't be sure but you might even have to change the Docker base image; maybe the standard Ubuntu xenial images doesn't work on the Pi. In any case, it's not an issue with this image -> closing

closed 2019-10-19 by marcelstoer

chathurangawijetunge, 2019-10-21:
no i get 
/opt:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/opt/nodemcu-firmware/tools/toolchains/esp8266-linux-x86_64-20181106.0/bin
/opt/nodemcu-firmware/tools/toolchains/esp8266-linux-x86_64-20181106.0/bin/xtensa-lx106-elf-ar: 1: /opt/nodemcu-firmware/tools/toolchains/esp8266-linux-x86_64-20181106.0/bin/xtensa-lx106-elf-ar: Syntax error: "(" unexpected
Makefile:334: recipe for target '/opt/nodemcu-firmware/sdk/.pruned-3.0-e4434aa' failed
make: *** [/opt/nodemcu-firmware/sdk/.pruned-3.0-e4434aa] Error 2
make: Leaving directory '/opt/nodemcu-firmware'

HHHartmann, 2019-10-21:
You are on docker? 
Start docker with`bash` instead of build. You will get a Linux prompt where you can build just the luac.cross as Terry described above. No need for xtensa then.

chathurangawijetunge, 2019-10-21:
im on RPI i have install docker and "hello world" is working fine
but when i run
docker run --rm -ti -v `pwd`:/opt/nodemcu-firmware marcelstoer/nodemcu-build build
i get 
standard_init_linux.go:211: exec user process caused "exec format error"

when i do 
# If you want to tinker with this Dockerfile on your machine do as follows:
# - git clone https://github.com/marcelstoer/docker-nodemcu-build
# - cd docker-nodemcu-build
# - vim Dockerfile
# - docker build -t docker-nodemcu-build .
# - cd <nodemcu-firmware>
# - docker run --rm -ti -v `pwd`:/opt/nodemcu-firmware docker-nodemcu-build build

i get 

/opt:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/opt/nodemcu-firmware/tools/toolchains/esp8266-linux-x86_64-20181106.0/bin
/opt/nodemcu-firmware/tools/toolchains/esp8266-linux-x86_64-20181106.0/bin/xtensa-lx106-elf-ar: 1: /opt/nodemcu-firmware/tools/toolchains/esp8266-linux-x86_64-20181106.0/bin/xtensa-lx106-elf-ar: Syntax error: "(" unexpected
Makefile:334: recipe for target '/opt/nodemcu-firmware/sdk/.pruned-3.0-e4434aa' failed
make: *** [/opt/nodemcu-firmware/sdk/.pruned-3.0-e4434aa] Error 2
make: Leaving directory '/opt/nodemcu-firmware'

HHHartmann, 2019-10-21:
Use `docker run --rm -ti -v pwd:/opt/nodemcu-firmware docker-nodemcu-build bash` to start a shell in the docker image and then build the luac.cross binaty as Terry detailed in the other bug over at nodemcu-firmware.
Leave the docker image with `exit`
After that building LFS should work with your command above:
`docker run --rm -ti -v pwd:/opt/nodemcu-firmware -v /home/pi/scw:/opt/lua marcelstoer/nodemcu-build lfs-image`

chathurangawijetunge, 2019-10-21:
If no trubble please tell me how to build luac.cross
On docker shell as new to docker

chathurangawijetunge, 2019-10-21:
from 
docker run --rm -ti -v pwd:/opt/nodemcu-firmware docker-nodemcu-build 
i get

Welcome to the Docker image to build NodeMCU firmware and LFS images.

To build the firmware or an LFS image you need to add the respective command
to the line you just used.
Say you used

  docker run --rm -it -v C:/Users/ ... nodemcu-firmware:/opt/nodemcu-firmware marcelstoer/nodemcu-build

now use

  docker run --rm -it -v C:/Users/ ... nodemcu-firmware:/opt/nodemcu-firmware marcelstoer/nodemcu-build build

Following commands are available:

build           builds your firmware, either for ESP32 or for ESP8266

configure-esp32 start the SDK configuration for ESP32

lfs-image       create an LFS image
                To do this you must tell docker where your lua files are stored.
                Add another '-v c:/lua_location:opt/lua' to the commandline.
                All lua files stored in that location will be added recursively
                to the LFS image.
				See tutorial below to see how to add selected files only.

For more Information please see https://hub.docker.com/r/marcelstoer/nodemcu-build

But with
docker run --rm -ti -v /home/pi/nodemcu-firmware:/opt/nodemcu-firmware marcelstoer/nodemcu-build build

i get
standard_init_linux.go:211: exec user process caused "exec format error"