marcelstoer/docker-nodemcu-build #82: Explain output filenames in Readme

mk-pmb, 2020-01-02:
Hi! Could you please explain the output filenames, and also the `IMAGE_NAME` option? I assume that

* `0x00000.bin` is just the firmware bytecode,
* `0x10000.bin` is just the SPIFFS,
* `nodemcu_float_master_yyyymmdd-hhmm.bin` is the firmware image I should flash, and
* `nodemcu_float_master_yyyymmdd-hhmm.map` are debug symbols,

but assuming is not knowing. :-)

marcelstoer, 2020-01-02:
* `0x00000.bin` is just the firmware bytecode,
* `0x10000.bin` is just the SPIFFS,

Yes, they are what the firmware project `make` produces. More precisely, this project's build scripts reproduce what the firmware CI build does.

* `nodemcu_float_master_yyyymmdd-hhmm.bin` is the firmware image I should flash, and

Yes -> https://github.com/marcelstoer/docker-nodemcu-build#output. Also take note of the `IMAGE_NAME` option as per https://github.com/marcelstoer/docker-nodemcu-build#options.

* `nodemcu_float_master_yyyymmdd-hhmm.map` are debug symbols,

It contains the relative offsets of functions.

Please also consider the "Support" section in the README, thanks.

closed 2020-01-02 by marcelstoer

mk-pmb, 2020-01-03:
Indeed I did read the "output" section. When I noticed it doesn't mention the `IMAGE_NAME` option yet, I considered that a docs bug, so I filed this issue, as suggested in the "support" section.

Anyway, thanks for clarifying the missing parts! Now I can prepare a PR to help improve the docs.

HHHartmann, 2020-01-03:
> * `0x10000.bin` is just the SPIFFS,

Ate you sure about this?
My SPIFFS never gets overwritten when I flash an image although the 0x10000.bin is included in the image. And there is #5