marcelstoer/docker-nodemcu-build #86: Bullding for ESP32 fails: setuptools missing

NightDragon1, 2020-01-20:
Hi! Just tried to make a build for esp32.
I get the following issue:

```
user001@hydra ~/nodemcu/nodemcu-esp32 $ docker run --rm -it -v /home/user001/nodemcu/nodemcu-esp32:/opt/nodemcu-firmware marcelstoer/nodemcu-build build
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
WARNING: The directory '/root/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Collecting setuptools
  Downloading https://files.pythonhosted.org/packages/f9/d3/955738b20d3832dfa3cd3d9b07e29a8162edb480bf988332f5e6e48ca444/setuptools-44.0.0-py2.py3-none-any.whl (583kB)
     |################################| 583kB 1.3MB/s
Requirement already satisfied: pyserial>=3.0 in /usr/lib/python2.7/dist-packages (from -r /opt/nodemcu-firmware/sdk/esp32-esp-idf/requirements.txt (line 8)) (3.0.1)
Collecting future>=0.15.2
  Downloading https://files.pythonhosted.org/packages/45/0b/38b06fd9b92dc2b68d58b75f900e97884c45bedd2ff83203d933cf5851c9/future-0.18.2.tar.gz (829kB)
     |################################| 829kB 705kB/s
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-6YPUXL/future/setup.py'"'"'; __file__='"'"'/tmp/pip-install-6YPUXL/future/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-6YPUXL/future/pip-egg-info
         cwd: /tmp/pip-install-6YPUXL/future/
    Complete output (3 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    ImportError: No module named setuptools
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

What am I missing?

Kritnich, 2020-02-01:
@NightDragon1 I was stuck with the same error but managed to get around it by entering the container and manually installing setuptools before running build.

Instead of
``docker run --rm -ti -v `pwd`:/opt/nodemcu-firmware marcelstoer/nodemcu-build build``
do
``docker run --rm -ti -v `pwd`:/opt/nodemcu-firmware marcelstoer/nodemcu-build /bin/bash``
then from inside the container do `pip install setuptools` followed by `build` to build the firmware.

marcelstoer, 2020-02-01:
That's an interesting observation, thanks. The build _does_ run a pip install for the all the modules the ESP-IDF requires as per 

https://github.com/marcelstoer/docker-nodemcu-build/blob/21aaeb1adfb796a10db53b523952dea99df145ec/build-esp32#L74

I'm wondering why it's not installing setuptools then.

Kritnich, 2020-02-01:
I believe I have found the problem, take a look: https://github.com/pypa/pip/issues/25

Either way, by removing setuptools from the `requirements.txt` and instead adding a line to `/opt/build-esp32` to install it separately beforehand it works.

marcelstoer, 2020-02-02:
Python isn't my home turf but I still don't get it - besides, I can't reproduce the problem. Why isn't everybody else having this issue?
Are both of you using the latest version of the Docker image (ensure by running `docker pull marcelstoer/nodemcu-build`) and the NodeMCU `dev-esp32` branch?
In the upstream ESP-IDF `requirements.txt` between v3.3.1 and master is only marginally different.

marcelstoer, 2020-02-03:
Whatever the root cause (or my lack of understanding thereof) it's easy enough to just add that line to the ESP32 build script.

closed 2020-02-03 by marcelstoer

mightymos, 2022-06-13:
I'm getting very similar setuptools error messages on Ubuntu 22.04 LTS (Jammy) (x86_64) and Raspberry Pi 4b (Armbian aarch64) for the docker image.
So my original goal was getting docker working on arm64, then build nodemcu, and finally for flashing hardware connected to the RPI4b.

When I try to explicitly install setuptools by Dockerfile I get this error:
```
Collecting setuptools
  Downloading https://files.pythonhosted.org/packages/0c/f0/f11edc0efc7b00e26a28225621ce7584636bb3a76b1862829d42bc146fbf/setuptools-62.4.0.tar.gz (2.6MB)
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "setuptools/__init__.py", line 8, in <module>
        import _distutils_hack.override  # noqa: F401
      File "_distutils_hack/override.py", line 1, in <module>
        __import__('_distutils_hack').do_override()
      File "_distutils_hack/__init__.py", line 72, in do_override
        ensure_local_distutils()
      File "_distutils_hack/__init__.py", line 55, in ensure_local_distutils
        importlib.import_module('distutils')
      File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
        __import__(name)
    AttributeError: DistutilsMetaFinder instance has no attribute 'find_module'
```
I'm wondering if there would be an openness to trying a PR for a newer version of Ubuntu (with newer pip, etc.) if building nodemcu seems to work?

commit 4e0af2eb (2020-02-03): Install setuptools for ESP32 explicitly

Fixes #86