marcelstoer/docker-nodemcu-build #87: USER_PROLOG

rvalle, 2020-03-16:
Hi!

I have integrated this builder image in our gitlab-ci process

I need to add information about out build, which I believe is done overriding USER_PROLOG

I am trying to set it up in passed environment parameter but does its not working, neither is setting it into app/include/user_version.h 

Is it possible? How?

rvalle, 2020-03-16:
in fact the BRANCH is also shown as HEAD, because of the checkout that gitlab-ci does.

marcelstoer, 2020-03-16:
The branch is detected automatically for both [ESP8266](https://github.com/marcelstoer/docker-nodemcu-build/blob/master/build-esp8266#L18) and [ESP32](https://github.com/marcelstoer/docker-nodemcu-build/blob/master/build-esp32#L21) with `$(git rev-parse --abbrev-ref HEAD | sed -r 's/[\/\\]+/_/g')`.

The env variable `USER_PROLOG` is exported by the build script before `make` is invoked at https://github.com/marcelstoer/docker-nodemcu-build/blob/master/build-esp8266#L11. If you want to control its value I currently see two options:
- build your own Docker image base on a forked and modified version of these scripts
- submit a PR that allows controlling `USER_PROLOG` through an env variable passed to `docker run`

closed 2020-03-16 by marcelstoer