marcelstoer/docker-nodemcu-build #90: Upgrading to Docker 2.3.0.2 Breaks NodeMCU-Build?

jmd13391, 2020-05-12:
Windows 10 -- Docker -- Current NodeMCU Dev Repo - All is well, Life is good...

Then...

I made the mistake of updating Docker to latest version 2.3.0.2 (45183)

Now, all hell broke loose... Running the following command (previously working)...

docker run --rm -it -e "IMAGE_NAME=test1" -v "//c/Users/Joe/Google Drive/nodemcu-firmware":/opt/nodemcu-firmware marcelstoer/nodemcu-build build

Throws the following error (yes, I have docker shared files configured to share C:\):

fatal: Not a git repository (or any parent up to mount point /opt/nodemcu-firmware)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).

Tried re-installing Docker (2.3.0.2) & Git for Windows (v 2.26.2) -- No joy. :-(

BTW/FYI... I believe you have transposed the "-it" docker run parm in just about all docs associated with docker-nodemcu-build. Docs show the parm as "-ti" and I think it should be "-it"?

marcelstoer, 2020-05-12:
> Upgrading to Docker 2.3.0.2 Breaks NodeMCU-Build?

Nope, not here. Looks like a user error but I can't say which one.

> Docs show the parm as "-ti" and I think it should be "-it"?

Run `docker run --help`. Those are two independent switches; could just as well write `-i -t`.

jmd13391, 2020-05-12:
> Nope, not here. Looks like a user error but I can't say which one.

Rats... I was hoping for a "you just need to..." fix. Literally, that docker run command has been working perfectly for well over a year. After a simple "OK" click to the most recent "Upgrade Now?" docker popup yesterday and BAM... it now throws an error. I'm stumped... and of course, it's right in front of a crushing software delivery deadline. :-(

> Those are two independent switches; could just as well write -i -t.

Ah... Duh... LOL Thanks!

jmd13391, 2020-05-12:
@marcelstoer, Anything jump out at you as being an issue here?

C:\Users\Joe>git clone --recurse-submodules -b dev https://github.com/nodemcu/nodemcu-firmware.git
Cloning into 'nodemcu-firmware'...
remote: Enumerating objects: 5, done.
remote: Counting objects: 100% (5/5), done.
remote: Compressing objects: 100% (5/5), done.
remote: Total 20056 (delta 0), reused 0 (delta 0), pack-reused 20051
Receiving objects: 100% (20056/20056), 108.26 MiB | 14.90 MiB/s, done.
Resolving deltas: 100% (13610/13610), done.
Updating files: 100% (1011/1011), done.
Submodule 'app/u8g2lib/u8g2' (https://github.com/olikraus/U8g2_Arduino.git) registered for path 'app/u8g2lib/u8g2'
Submodule 'app/ucglib/ucg' (https://github.com/olikraus/Ucglib_Arduino.git) registered for path 'app/ucglib/ucg'
Cloning into 'C:/Users/Joe/nodemcu-firmware/app/u8g2lib/u8g2'...
remote: Enumerating objects: 1148, done.
remote: Counting objects: 100% (1148/1148), done.
remote: Compressing objects: 100% (188/188), done.
remote: Total 10010 (delta 578), reused 1057 (delta 501), pack-reused 8862
Receiving objects: 100% (10010/10010), 47.55 MiB | 10.48 MiB/s, done.
Resolving deltas: 100% (5406/5406), done.
Cloning into 'C:/Users/Joe/nodemcu-firmware/app/ucglib/ucg'...
remote: Enumerating objects: 376, done.
remote: Total 376 (delta 0), reused 0 (delta 0), pack-reused 376
Receiving objects: 100% (376/376), 5.18 MiB | 7.02 MiB/s, done.
Resolving deltas: 100% (213/213), done.
Submodule path 'app/u8g2lib/u8g2': checked out '2ee84c8f14adaa8fd1ebfe091c4de348c5474b18'
Submodule path 'app/ucglib/ucg': checked out 'e21641a6c1ddb0e71f7b9e01501fa739786c68b1'

C:\Users\Joe>docker pull marcelstoer/nodemcu-build
Using default tag: latest
latest: Pulling from marcelstoer/nodemcu-build
Digest: sha256:27acd488ff51f1b3c0abb4a55497d6746b0f32afc626ed21cef32f21562396fe
Status: Image is up to date for marcelstoer/nodemcu-build:latest
docker.io/marcelstoer/nodemcu-build:latest

C:\Users\Joe>docker image ls --all
REPOSITORY                  TAG                 IMAGE ID            CREATED             SIZE
marcelstoer/nodemcu-build   latest              913000b4b968        8 weeks ago         464MB
hello-world                 latest              bf756fb1ae65        4 months ago        13.3kB

C:\Users\Joe>docker run --rm -ti -e "IMAGE_NAME=test1" -v "//c/Users/Joe/nodemcu-firmware":/opt/nodemcu-firmware marcelstoer/nodemcu-build build
fatal: Not a git repository (or any parent up to mount point /opt/nodemcu-firmware)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).

C:\Users\Joe>

jmd13391, 2020-05-12:
@marcelstoer  -- Something is seriously broken with the docker-nodemcu-build when running under Windows. I originally thought it was caused by me upgrading to the latest version of Docker but now I'm not so sure. To eliminate the root cause being something squirrely/corrupt on my active development system, I performed a fresh install of Docker, Git, nodemcu-firmware repo, etc. on a new, completely vanilla Windows 10 system and was presented with the exact same error.

My development project is current hard-down and I am at a loss as to how to proceed here. Any assistance you can provide would be greatly appreciated. I am stuck.  :-(

jmd13391, 2020-05-12:
@marcelstoer -- Strange... changing the [-v "//c/Users/Joe/nodemcu-firmware"] to [-v "c:/Users/Joe/nodemcu-firmware"] resolves the issue.

closed 2020-05-12 by jmd13391