marcelstoer/docker-nodemcu-build #91: Add support for building LFS or SPIFFS images?

nwf, 2020-05-24:
It would be useful, I think, if we could pick at least in-tree `lua_modules` directly into a LFS image as part of the firmware bundle.  This would lower the friction of jettisoning some "somewhat hairy" C modules in favor of Lua+C hybrids as I've been trying to drive us to (e.g. https://github.com/nodemcu/nodemcu-firmware/pull/2819 and now https://github.com/nodemcu/nodemcu-firmware/issues/3126).

However, I understand if there's hesitation to do LFS builds because people might want to add their own modules, and that opens "now I need to let people upload files" which is always a little hair-raising on the web.  So...

Maybe instead of LFS, landing selected `lua_modules` in SPIFFS would still be useful.  They could still be `require`-d as is, at the cost of increased memory consumption, and given @TerryE's proposal in https://github.com/nodemcu/nodemcu-firmware/issues/2917, having Lua files in SPIFFS would eventually let us tell people how to compile their LFS images on-module.

marcelstoer, 2020-05-25:
I doubt I fully understand the issue as LFS has been supported for a long time thanks to @HHHartmann : https://github.com/marcelstoer/docker-nodemcu-build#create-an-lfs-image-for-esp8266

nwf, 2020-05-25:
Argh, I was apparently not completely awake when I filed this one.  I meant for this to be about the web builder. -_-

closed 2020-05-26 by nwf

marcelstoer, 2020-05-28:
The correct list would be https://github.com/marcelstoer/nodemcu-custom-build/issues (open-source backend of the cloud builder).