marcelstoer/docker-nodemcu-build #93: pulsecnt module seems to be "missing" in branch esp32-dev

n07-5l4y3r, 2020-09-15:
So today I tryed my best to get the Pulse Counter working on my ESP32 but encountered some quite unique issues.

Output Log:
[image]
<details>
  <summary>Full Log</summary>
  
  I also find it interesting that the chip is doing something with wifi but the firmware shouldn't include this module.
  (Re-Requesting build's also didn't do the trick)
  
  ```
 node.restart()
> node.restart()
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6736
ho 0 tail 12 room 4
load:0x40078000,len:11116
load:0x40080400,len:5940
entry 0x40080700
[0;32mI (73) boot: Chip Revision: 1[0m
[0;32mI (73) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m
[0;32mI (41) boot: ESP-IDF v3.3-beta2-887-g9e70825d1 2nd stage bootloader[0m
[0;32mI (41) boot: compile time 17:12:49[0m
[0;32mI (41) boot: Enabling RNG early entropy source...[0m
[0;32mI (48) boot: SPI Speed      : 40MHz[0m
[0;32mI (52) boot: SPI Mode       : DIO[0m
[0;32mI (56) boot: SPI Flash Size : 4MB[0m
[0;32mI (60) boot: Partition Table:[0m
[0;32mI (64) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (86) boot:  2 factory          factory app      00 00 00010000 00180000[0m
[0;32mI (93) boot:  3 lfs              unknown          c2 01 00190000 00010000[0m
[0;32mI (101) boot:  4 nodemcuspiffs    unknown          c2 00 001a0000 00070000[0m
[0;32mI (108) boot: End of partition table[0m
[0;32mI (113) boot_comm: chip revision: 1, min. application chip revision: 0[0m
[0;32mI (120) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x24c30 (150576) map[0m
[0;32mI (180) esp_image: segment 1: paddr=0x00034c58 vaddr=0x3ffb0000 size=0x035e8 ( 13800) load[0m
[0;32mI (185) esp_image: segment 2: paddr=0x00038248 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (187) esp_image: segment 3: paddr=0x00038650 vaddr=0x40080400 size=0x079c0 ( 31168) load[0m
[0;32mI (207) esp_image: segment 4: paddr=0x00040018 vaddr=0x400d0018 size=0xa0478 (656504) map[0m
[0;32mI (430) esp_image: segment 5: paddr=0x000e0498 vaddr=0x40087dc0 size=0x0d5bc ( 54716) load[0m
[0;32mI (451) esp_image: segment 6: paddr=0x000eda5c vaddr=0x400c0000 size=0x00064 (   100) load[0m
[0;32mI (465) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (465) boot: Disabling RNG early entropy source...[0m
[0;32mI (465) cpu_start: Pro cpu up.[0m
[0;32mI (469) cpu_start: Application information:[0m
[0;32mI (474) cpu_start: Project name:     NodeMCU[0m
[0;32mI (479) cpu_start: App version:      dd3b217-dirty[0m
[0;32mI (484) cpu_start: Compile time:     Sep 15 2020 17:12:52[0m
[0;32mI (490) cpu_start: ELF file SHA256:  8de9dabc96a3e962...[0m
[0;32mI (496) cpu_start: ESP-IDF:          v3.3-beta2-887-g9e70825d1[0m
[0;32mI (503) cpu_start: Starting app cpu, entry point is 0x400812e8[0m
[0;32mI (489) cpu_start: App cpu up.[0m
[0;32mI (513) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (520) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (526) heap_init: At 3FFB9CC8 len 00026338 (152 KiB): DRAM[0m
[0;32mI (533) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (539) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (545) heap_init: At 4009537C len 0000AC84 (43 KiB): IRAM[0m
[0;32mI (551) cpu_start: Pro cpu start user code[0m
[0;32mI (234) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m

Mounting flash filesystem...
No LFS image loaded
I (395) wifi:wifi driver task: 3ffc4430, prio:23, stack:3584, core=0
[0;32mI (395) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (395) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
I (425) wifi:wifi firmware version: 44aa95c
I (425) wifi:config NVS flash: enabled
I (425) wifi:config nano formating: disabled
I (425) wifi:Init dynamic tx buffer num: 32
I (435) wifi:Init data frame dynamic rx buffer num: 32
I (435) wifi:Init management frame dynamic rx buffer num: 32
I (445) wifi:Init management short buffer num: 32
I (445) wifi:Init static rx buffer size: 1600
I (455) wifi:Init static rx buffer num: 10
I (455) wifi:Init dynamic rx buffer num: 32

NodeMCU ESP32 built with Docker provided by frightanic.com
	branch: dev-esp32
	commit: f4887bf134235c05e6c9b2efad370e6d5018f91a
	SSL: true
	modules: -
 built on nodemcu-build.com provided by frightanic.com
	branch: dev-esp32
	commit: dd3b217df201fde00fcf84be1d485d555d433bed
	SSL: false
	modules: file,gpio,node,pulsecnt,tmr,uart
 build 2020-09-15-17-10-55 powered by Lua 5.1.4 on ESP-IDF v3.3-beta1-1391-g9e70825d1 on SDK IDF
Hello World!
file
romtable: 0x3f4042a4
gpio
romtable: 0x3f404530
node
romtable: 0x3f403fe8
pulsecnt
nil
tmr
romtable: 0x3f404b34
uart
romtable: 0x3f409064
  ```
  
</details>

Code in question:
[image]

Firmware used:
[image]
[image]
[image]
* [integer](https://mega.nz/file/FAkyxYJD#Pj7-6HKv1Ynp6l99xnlopKlScj2tGk6qgaI82flRfek)
* [float](https://mega.nz/file/VY0zmYwL#sK-fm7Y0g_tK5DP-yP1qjfNEarBxzJ_UjLptWE9IcRk)
(tested both)

Tool-Chain used:
https://nodemcu-build.com/index.php
http://chilipeppr.com/esp32
https://github.com/marcelstoer/nodemcu-pyflasher/releases/download/v4.0/NodeMCU-PyFlasher-4.0-x64.exe

Flash Log:
[image]

I am unsure how to act further therefore im opening this issue (which is btw my first so please excuse me for the bad writeup)

marcelstoer, 2020-09-15:
This...does not make any sense at all.

```
NodeMCU ESP32 built with Docker provided by frightanic.com
  branch: dev-esp32
  commit: f4887bf134235c05e6c9b2efad370e6d5018f91a
  SSL: true
  modules: -
built on nodemcu-build.com provided by frightanic.com
  branch: dev-esp32
  commit: dd3b217df201fde00fcf84be1d485d555d433bed
  SSL: false
  modules: file,gpio,node,pulsecnt,tmr,uart
build 2020-09-15-17-10-55 powered by Lua 5.1.4 on ESP-IDF v3.3-beta1-1391-g9e70825d1 on SDK IDF
```

You _either_ flash the firmware from the cloud builder - what you did according to the screen shot - or the one you built yourself with Docker - which you also seem to have done. I don't understand how you could get a mix therefore? And what's with ChilliPeppr?

Here's what I would do:
- build a binary on the cloud builder _using only the default modules_
- flash it
- confirm you get a clean boot message
- connect to your WiFi

If all is well then continue by building a new firmware with the modules you really need and repeat.

n07-5l4y3r, 2020-09-15:
The firmware used in those screenshots is one from the [cloud-builder.](https://nodemcu-build.com/index.php) I havent attempted building it myself.

The script in the end shall have just one ability: counting a single channel and returning this measurement through rs232.
Therefore no WiFi connection is needed as USB/Serial via UART is enough for this case.
So I obviously need the `pulsecnt` module: which seems undefined according [this](https://user-images.githubusercontent.com/52784025/93248902-cb840680-f790-11ea-8d0e-f94b315e802e.png) screen tho, which right now is my only problem.

> And what's with ChilliPeppr?
Thats just the "IDE" im using for uploading/executing the script.

So again what I did so far:
 * forged a binary using the cloud builder
 * flashed it
 * got boot
 * uploaded hello world script
 * got expected reply
 * uploaded this script [image]
 * also got expected reply EXCEPT `pulsecnt == nil`

----

I am requesting a build without pulsecnt right now which will prolly execute and run just as good but obviously will also not have the pulse counter module. (next comment upcomming)

n07-5l4y3r, 2020-09-15:
As promised (and expected):

<details>
  <summary>Mail</summary>

  ```
Your NodeMCU custom build finished successfully. You may now download the firmware:
- float: https://nodemcu-build.com/builds/nodemcu-dev-esp32-7-modules-2020-09-15-19-44-47-float.bin
- integer: https://nodemcu-build.com/builds/nodemcu-dev-esp32-7-modules-2020-09-15-19-44-47-integer.bin

This was built against the dev-esp32 branch and includes the following modules: file, gpio, net, node, tmr, uart, wifi.
The files are guaranteed to be available for download for 24h.

I invite you to try my standalone desktop tool to flash firmware binaries: https://github.com/marcelstoer/nodemcu-pyflasher

Learn how to flash the firmware: https://nodemcu.readthedocs.io/en/dev-esp32/en/flash/. If you're upgrading from a previous version you should carefully study the "Upgrading Firmware" instructions on that page.

Please help keep this service alive: http://nodemcu-build.com/faq.php#support
  ```

</details>

<details>
  <summary>Flash-Log</summary>

  ```
Command: esptool.py --port COM3 --baud 115200 --after no_reset write_flash --flash_mode dio 0x00000 F:\Users\user\Downloads\nodemcu-dev-esp32-7-modules-2020-09-15-19-44-47-float.bin --erase-all

esptool.py v2.6
Serial port COM3
Connecting......
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 24:62:ab:ff:8e:64
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Erasing flash (this may take a while)...
Chip erase completed successfully in 7.8s
Compressed 985840 bytes to 615676...
Wrote 985840 bytes (615676 compressed) at 0x00000000 in 54.6 seconds (effective 144.4 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.

Firmware successfully flashed. Unplug/replug or reset device 
to switch back to normal boot mode.
  ```

</details>

</details>

<details>
  <summary>stdout</summary>

  ```
node.restart()
node.restart()
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
[... 155 lines cut ...]
I (439) wifi:Init management frame dynamic rx buffer num: 32
I (449) wifi:Init management short buffer num: 32
I (449) wifi:Init static rx buffer size: 1600
I (459) wifi:Init static rx buffer num: 10
I (459) wifi:Init dynamic rx buffer num: 32

NodeMCU ESP32 built with Docker provided by frightanic.com
	branch: dev-esp32
	commit: f4887bf134235c05e6c9b2efad370e6d5018f91a
	SSL: true
	modules: -
 built on nodemcu-build.com provided by frightanic.com
	branch: dev-esp32
	commit: dd3b217df201fde00fcf84be1d485d555d433bed
	SSL: false
	modules: file,gpio,net,node,tmr,uart,wifi
 build 2020-09-15-19-44-47 powered by Lua 5.1.4 on ESP-IDF v3.3-beta1-1391-g9e70825d1 on SDK IDF
Hello World!
  ```

</details>

------

Next Comment will be the exact same but with the "Pulse Counter" module.

n07-5l4y3r, 2020-09-15:
<details>
  <summary>Mail</summary>
  
  ```
Your NodeMCU custom build finished successfully. You may now download the firmware:
- float: https://nodemcu-build.com/builds/nodemcu-dev-esp32-8-modules-2020-09-15-20-02-16-float.bin
- integer: https://nodemcu-build.com/builds/nodemcu-dev-esp32-8-modules-2020-09-15-20-02-16-integer.bin

This was built against the dev-esp32 branch and includes the following modules: file, gpio, net, node, pulsecnt, tmr, uart, wifi.
The files are guaranteed to be available for download for 24h.

I invite you to try my standalone desktop tool to flash firmware binaries: https://github.com/marcelstoer/nodemcu-pyflasher

Learn how to flash the firmware: https://nodemcu.readthedocs.io/en/dev-esp32/en/flash/. If you're upgrading from a previous version you should carefully study the "Upgrading Firmware" instructions on that page.

Please help keep this service alive: http://nodemcu-build.com/faq.php#support
  ```
  
</details>

<details>
  <summary>Flash-Log</summary>
  
  ```
Command: esptool.py --port COM3 --baud 115200 --after no_reset write_flash --flash_mode dio 0x00000 F:\Users\user\Downloads\nodemcu-dev-esp32-8-modules-2020-09-15-20-02-16-float.bin --erase-all

esptool.py v2.6
Serial port COM3
Connecting.....
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 24:62:ab:ff:8e:64
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Erasing flash (this may take a while)...
Chip erase completed successfully in 7.8s
Compressed 985856 bytes to 615704...
Wrote 985856 bytes (615704 compressed) at 0x00000000 in 54.6 seconds (effective 144.3 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.

Firmware successfully flashed. Unplug/replug or reset device 
to switch back to normal boot mode.
  ```
  
</details>

<details>
  <summary>stdout</summary>
  
  ```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6736
ho 0 tail 12 room 4
load:0x40078000,len:11116
load:0x40080400,len:5940
entry 0x40080700
[0;32mI (73) boot: Chip Revision: 1[0m
[... 182 lines cut ...]
romtable: 0x3f404a70
pulsecnt module loaded?
nil
  ```
  
</details>

<details>
  <summary>Script</summary>
  
  ```
print("Hello World!")

print("gpio module loaded?")
print(gpio)

print("pulsecnt module loaded?")
print(pulsecnt)
  ```
  
</details>

----

same problem sadly 😞

marcelstoer, 2020-09-15:
I'm a little less confused now. This hit me out of the blue, sorry.

The weird boot message appears to be a bug - never seen anything like that. I'm tracking this at https://github.com/marcelstoer/nodemcu-custom-build/issues/39.

The missing pulse counter module is related to an [upstream change in the firmware Kconfig](https://github.com/nodemcu/nodemcu-firmware/pull/3135) that wasn't properly reflected in the cloud builder. This is already fixed.

closed 2020-09-15 by marcelstoer