marcelstoer/docker-nodemcu-build #95: ESP8266 build fails on WGET - no such file

pfurbish, 2020-10-03:
Sorry to bother but I'm getting the following error.
Below is a partial output from the build but I can send all if needed.
WGET fails...

Creating image for ESP8266...
make[1]: warning:  Clock skew detected.  Your build may be incomplete.
make[1]: Leaving directory '/opt/nodemcu-firmware/app'
make[1]: Entering directory '/opt/nodemcu-firmware/nodemcu-firmware'
../Makefile:348: warning: overriding recipe for target 'clean'
Makefile:348: warning: ignoring old recipe for target 'clean'
../Makefile:353: warning: overriding recipe for target 'clobber'
Makefile:353: warning: ignoring old recipe for target 'clobber'
../Makefile:357: warning: overriding recipe for target 'flash'
Makefile:357: warning: ignoring old recipe for target 'flash'
../Makefile:363: warning: overriding recipe for target 'flash512k'
Makefile:363: warning: ignoring old recipe for target 'flash512k'
../Makefile:366: warning: overriding recipe for target 'flash4m'
Makefile:366: warning: ignoring old recipe for target 'flash4m'
../Makefile:369: warning: overriding recipe for target 'flash1m-dout'
Makefile:369: warning: ignoring old recipe for target 'flash1m-dout'
../Makefile:376: warning: overriding recipe for target 'flashinternal'
Makefile:374: warning: ignoring old recipe for target 'flashinternal'
../Makefile:380: warning: overriding recipe for target '.subdirs'
Makefile:380: warning: ignoring old recipe for target '.subdirs'
../Makefile:393: warning: overriding recipe for target 'spiffs-image-remove'
Makefile:393: warning: ignoring old recipe for target 'spiffs-image-remove'
../Makefile:398: warning: overriding recipe for target 'spiffs-image'
Makefile:398: warning: ignoring old recipe for target 'spiffs-image'
../Makefile:412: warning: overriding recipe for target 'pre_build'
Makefile:412: warning: ignoring old recipe for target 'pre_build'
../Makefile:418: warning: overriding recipe for target 'buildinfo'
Makefile:418: warning: ignoring old recipe for target 'buildinfo'
WGET nodemcu-firmware/cache/toolchain-esp8266-linux-x86_64-20190731.0.tar.xz
/opt/nodemcu-firmware/nodemcu-firmware/cache/toolchain-esp8266-linux-x86_64-20190731.0.tar.xz: **No such file or directory**
Makefile:301: recipe for target '/opt/nodemcu-firmware/nodemcu-firmware/cache/toolchain-esp8266-linux-x86_64-20190731.0.tar.xz' failed
make[1]: *** [/opt/nodemcu-firmware/nodemcu-firmware/cache/toolchain-esp8266-linux-x86_64-20190731.0.tar.xz] Error 1
make[1]: Leaving directory '/opt/nodemcu-firmware/nodemcu-firmware'
Makefile:380: recipe for target '.subdirs' failed
make: *** [.subdirs] Error 2
make: Leaving directory '/opt/nodemcu-firmware'
PS C:\Users\paul\nodemcu-firmware>

My environment:
Docker v19.03.13 on Windows 10 using WSL 2 - installed today
PowerShell 7.0.3
Image: docker pull marcelstoer/nodemcu-build
Git: git clone --recurse-submodules https://github.com/nodemcu/nodemcu-firmware.git
Build: docker run --rm -ti -v c:/Users/paul/nodemcu-firmware:/opt/nodemcu-firmware marcelstoer/nodemcu-build build
The same results after 3 attempts.

[image]

Thanks.

marcelstoer, 2020-10-04:
I don't see the part where wget is failing in your output. Try wiping the `cache` directory.

It should contain artifacts as follows

```
drwxr-xr-x 2 root root     4096 Oct  4 06:16 esptool
-rw-r--r-- 1 root root 21543188 Jul 31  2019 toolchain-esp8266-linux-x86_64-20190731.0.tar.xz
-rw-r--r-- 1 root root  6427787 Oct  4 06:16 v3.0.4.zip
```
Looks like it failed to store `v3.0.4.zip` (the Espressif SDK) under the correct name.

pfurbish, 2020-10-04:
I deleted my nodemcu-firmware folder then ran git and now the docker firmware build is working.
Thank you for your help.

closed 2020-10-04 by pfurbish