marcelstoer/docker-nodemcu-build #96: Updating nodeMCU firmware builds for SparkFun ESP32 Thing

Cyberfonic, 2021-01-05:
The ESP32 Thing has a 26MHz xtal and thus the cloud built firmware doesn't work correctly.  The recommended solution is to build using [Docker based build environment](https://hub.docker.com/r/marcelstoer/nodemcu-build/). It worked fine for me back in April 2020.

I have the need to update the firmware so I followed the instructions on [NodeMCU firmware page](https://nodemcu.readthedocs.io/en/dev-esp32/build/) but I ended up making a hash of it.  Mainly because my git expertise is abysmal. I ended up wiping my old build directory and starting afresh.

Would have preferred to have some notes on how to accomplish a refresh of the firmware build directory.  Any chance that the above page could include some instructions to that effect?

marcelstoer, 2021-01-05:
> the above page

Does this refer to the README of this project or the NodeMCU documentation?

Cyberfonic, 2021-01-05:
I'm referring specifically to the instructions on:  https://hub.docker.com/r/marcelstoer/nodemcu-build/

They are good for first time use. But due to my lack of experience with git, I was unable to correctly update the files.

marcelstoer, 2021-01-13:
I added a few notes with 34f99b0

closed 2021-01-13 by marcelstoer