marcelstoer/docker-nodemcu-build #97: Unable to build ESP8266 firmware: README out of date

sgily, 2021-04-18:
Hi,
   as preface, i'll say that i have no experience with dockers, so in part i have no idea what i'm doing. It is possible that i screwed up somewhere, although i simply followed the guidelines.

Issue 1: app directory no longer exists.
    Instructions say to do
        git checkout dev-esp32 - but the dev-esp32 branch does not have app/include directory, and user_config.h file can no longer be found there.

Issue 2: unable to complete docker build
~/nodemcu-firmware$ sudo docker run --rm -ti -v 'pwd':/opt/nodemcu-firmware marcelstoer/nodemcu-build build
fatal: Not a git repository (or any parent up to mount point /opt/nodemcu-firmware)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).

Changing URL to repository from "marcelstoer/nodemcu-build" to "marcelstoer/docker-nodemcu-build " yields:
~/nodemcu-firmware$ sudo docker run --rm -ti -v 'pwd':/opt/nodemcu-firmware marcelstoer/docker-nodemcu-build build
Unable to find image 'marcelstoer/docker-nodemcu-build:latest' locally
docker: Error response from daemon: pull access denied for marcelstoer/docker-nodemcu-build, repository does not exist or may require 'docker login': denied: requested access to the resource is denied.
See 'docker run --help'.

jyh29, 2021-05-19:
Not familiar with Docker, Only for compiling nodemcu-build Docker has been successfully installed on win10, and its relative hyper-v & WSL2 Ubuntu
also working correctly, but Docker compiling nodemcu-build is always wrong, with error message:
docker run --rm -it -v //c/Users/Administrator/Downloads/nodemcu-firmware-master:/opt/nodemcu-firmware marcelstoer/nodemcu-build build
fatal: Not a git repository (or any parent up to mount point /opt/nodemcu-firmware)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set)

This being as expected, because the all of articles have never mentioned about the settings of hyper-v or Ubuntu related configuration for the compiling nodemcu-build.
So I wander if with this compiling mode WSL2 Ubuntu may not need installed ?
Otherwise, how to set the download folder in Win10 for downloading the nodemcu-build ?
how to check or comfirm the right state of files in Ubuntu working directories ready for Docker build before ?
where to get the compiled nodemcu-firmware from Ubuntu working directory ?

marcelstoer, 2021-05-19:
> Issue 1: app directory no longer exists.

Unrelated to Docker or this image. If you want to build the image yourself please familiarize yourself with [ESP-IDF](https://github.com/espressif/esp-idf) and the [NodeMCU `dev-esp32` branch](https://nodemcu.readthedocs.io/en/dev-esp32/build/#build-options).

> Issue 2: unable to complete docker build

Just copy-paste the commands from the README as completely as possible. It's `` `pwd` `` and not `'pwd'`.

> Changing URL to repository from "marcelstoer/nodemcu-build" to "marcelstoer/docker-nodemcu-build"

Just checked, the README has no mention of that 2nd Docker image. Neither has the [description on Docker Hub](https://hub.docker.com/r/marcelstoer/nodemcu-build/). I don't know where you got that from.

closed 2021-05-19 by marcelstoer