marcelstoer/docker-nodemcu-build #98: How to check or comfirm the right state of files in Linux working directories ready for Docker build before

jyh29, 2021-05-19:
Not familiar with Docker, Only for compiling nodemcu-build Docker has been successfully installed on win10, and its relative hyper-v & WSL2 Ubuntu
also working correctly, but Docker compiling nodemcu-build is always wrong, with error message:
docker run --rm -it -v //c/Users/Administrator/Downloads/nodemcu-firmware-master:/opt/nodemcu-firmware marcelstoer/nodemcu-build build
fatal: Not a git repository (or any parent up to mount point /opt/nodemcu-firmware)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set)

This being as expected, because the all of articles have never mentioned about the settings of hyper-v or Ubuntu related configuration for the compiling nodemcu-build.
So I wander if with this compiling mode WSL2 Ubuntu may not need installed ?
Otherwise, how to set the download folder in Win10 for downloading the nodemcu-build ?
how to check or comfirm the right state of files in Ubuntu working directories ready for Docker build before ?
where to get the compiled nodemcu-firmware from Ubuntu working directory ?

marcelstoer, 2021-05-19:
Please see https://github.com/marcelstoer/docker-nodemcu-build/blob/master/README.md#notes-for-windows-users

`//c/Users/Administrator/Downloads/nodemcu-firmware-master` is not understood by Docker as a valid path to your NodeMCU source code. This is unrelated to this project but a general Docker volume mounting question. Please refer to https://github.com/marcelstoer/docker-nodemcu-build/blob/master/README.md#support.

closed 2021-05-19 by marcelstoer

jyh29, 2021-05-20:
Very thanks for the rapid answer !
If possible, please help me to confirm that hyper-v Ubuntu must be installed and 
[git clone --recurse-submodules https://github.com/nodemcu/nodemcu-firmware.git] must executed successfully in the Ubuntu before
[docker run --rm -it -v //c/Users/Administrator/Downloads/nodemcu-firmware-master:/opt/nodemcu-firmware marcelstoer/nodemcu-build build] in win10 Docker.
I searched about that over and over on Internet, but failed to find out about that.
Thank you again in advance

marcelstoer, 2021-05-20:
Sorry, can't help you with these general Docker on Win10 questions - I don't use Windows.