marcelstoer/docker-nodemcu-build #99: Failed in last step No way to seek for solution

jyh29, 2021-05-23:
EXTRACT cache/toolchain-esp8266-linux-x86_64-20190731.0.tar.xz
--2021-05-23 01:11:50--  https://github.com/espressif/esptool/archive/v2.6.tar.gz
Resolving github.com (github.com)... 52.74.223.119
Connecting to github.com (github.com)|52.74.223.119|:443... connected.
Unable to establish SSL connection.
Makefile:316: recipe for target '/opt/nodemcu-firmware/cache/esptool/v2.6.tar.gz' failed
make: *** [/opt/nodemcu-firmware/cache/esptool/v2.6.tar.gz] Error 1
make: Leaving directory '/opt/nodemcu-firmware'

marcelstoer, 2021-05-23:
I cannot reproduce this. Works for me as documented.

jyh29, 2021-05-25:
Very thank you !
I am a beginner to Nodemcu, and my English ability is very poor,  even not know the exact meaning of words 'I cannot reproduce this. Works for me as documented.'
I abandon to use docker to build the Nodemcu-firmware.
Thank you again !

closed 2022-01-02 by marcelstoer