marcelstoer/docker-nodemcu-build #102: Is this project still maintained?

signed-log, 2023-04-25:
Seeing that there hasn't been an update to DNB since 2021, I'd like to ask if this project is still maintained. If it's not anymore, I'd love to take it over/help maintain it

marcelstoer, 2023-04-25:
Thanks for reaching out. The project is still maintained as in "I'll fix bugs and accept PRs". However, the last version either just works or...nobody is using it anymore 😄 Hence, no new commits in quite a while.

The only thing that really bugs me about the current version is #101. So, I'd be gladly accepting your offer to help maintain this project.

marcelstoer, 2023-04-25:
Oh, interesting, didn't know they've been publishing IDF Docker images for years. Offering a single image certainly has its advantages for the user (maybe not for the maintainers).

- I wanted this image to use the same Ubuntu version as we use for the NodeMCU CI build on GitHub. That's currently not the case anymore; not good: https://github.com/nodemcu/nodemcu-firmware/blob/release/.github/workflows/build.yml#L23
- As NodeMCU for ESP32 pulls in the IDF as a Git submodule https://github.com/nodemcu/nodemcu-firmware/blob/dev-esp32/.gitmodules I see potential version conflicts it we were to use the IDF Docker image here.

signed-log, 2023-04-25:
>     I see potential version conflicts it we were to use the IDF Docker image here.

I would just find a way to bypass the submodule and use the built-in toolchain (this would mean just bypassing the Makefile and calling `idf.py` directly)

marcelstoer, 2023-04-25:
> this would mean just bypassing the Makefile and calling `idf.py` directly

From what I understand this is quite likely to fail or be instable. NodeMCU is strictly tied to the one specific IDF version it references through the Git submodule.

signed-log, 2023-04-25:
`v4.4.3` is available as an [image](https://hub.docker.com/layers/espressif/idf/v4.4.3/images/sha256-8fc51413055e31684e1eac763b65cadfb202e90ef51a4d95768c21962398d63d?context=explore)

signed-log, 2023-04-25:
Seeing the issue now, this is a non-starter yeah

marcelstoer, 2024-01-15:
Any further discussions should continue at #101.

closed 2024-01-15 by marcelstoer