marcelstoer/docker-nodemcu-build #103: configure-esp32 gives error related to python3 absence

adityachd, 2023-05-04:
Working on windows 11.
Using the following command gives me this error on windows
`docker run --rm -it -v C:/Users/eltxa/nodemcu-firmware:/opt/nodemcu-firmware marcelstoer/nodemcu-build configure-esp32`
   
`Entering 'components/qrcodegen/qrcodegen'
Entering 'components/u8g2/u8g2'
Entering 'components/ucg/ucg'
Entering 'sdk/esp32-esp-idf'
. /opt/nodemcu-firmware/sdk/esp32-esp-idf/export.sh && make "menuconfig"
Setting IDF_PATH to '/opt/nodemcu-firmware/sdk/esp32-esp-idf'
Detecting the Python interpreter
Checking "python" ...
Checking "python3" ...
/opt/nodemcu-firmware/sdk/esp32-esp-idf/tools/detect_python.sh: line 16: python3: command not found
Python 2.7.12
"python" has been detected
Adding ESP-IDF tools to PATH...
  File "/opt/nodemcu-firmware/sdk/esp32-esp-idf/tools/idf_tools.py", line 406
    info(f'Downloading {url}')
                            ^
SyntaxError: invalid syntax
Makefile:13: recipe for target 'menuconfig' failed`

Is there a workaround or steps I might be missing?

marcelstoer, 2023-05-04:
This is likely a consequence of #101 -> closing

closed 2023-05-04 by marcelstoer

adityachd, 2023-05-04:
Okay. 
Thanks for the quick response.