marcelstoer/docker-nodemcu-build #104: python3 not detect in the Docker image

MajorLee95, 2023-10-25:
I try to compile nodeMcu for an esp32. I am on Windows 10.

When i try the following command:

docker run --rm -it -v c:/Users/<userName>/nodemcu-firmware:/opt/nodemcu-firmware marcelstoer/nodemcu-build configure-esp32

Container starts but rapidly stop on the following error:

```
Entering 'components/qrcodegen/qrcodegen'
Entering 'components/u8g2/u8g2'
Entering 'components/ucg/ucg'
Entering 'sdk/esp32-esp-idf'
. /opt/nodemcu-firmware/sdk/esp32-esp-idf/export.sh && make "menuconfig"
Setting IDF_PATH to '/opt/nodemcu-firmware/sdk/esp32-esp-idf'
Detecting the Python interpreter
Checking "python" ...
Checking "python3" ...
/opt/nodemcu-firmware/sdk/esp32-esp-idf/tools/detect_python.sh: line 16: python3: command not found
Python 2.7.12
"python" has been detected
Adding ESP-IDF tools to PATH...
  File "/opt/nodemcu-firmware/sdk/esp32-esp-idf/tools/idf_tools.py", line 406
    info(f'Downloading {url}')
                            ^
SyntaxError: invalid syntax
Makefile:13: recipe for target 'menuconfig' failed
make: *** [menuconfig] Error 1
```
It seems there is no python3 in the container, and it detects python2 but i think that the line "info(f'Downloading {url}')" is clearly Python3's syntaxe.

Do you have any idea ? I ckeck in the container but i didn't find any trace of Python3...

wortkrieg, 2023-11-04:
+1 on this. I have the exact same problem. Tried on Debian/amd64 & macOS/arm64. Also, I tried to modify the Dockerfile to use python3 & pip3 but with no luck, it breaks at exactly the same line. Output is:

`. /opt/nodemcu-firmware/sdk/esp32-esp-idf/export.sh && make "menuconfig"
Setting IDF_PATH to '/opt/nodemcu-firmware/sdk/esp32-esp-idf'
Detecting the Python interpreter
Checking "python" ...
Checking "python3" ...
Python 3.5.2
"python3" has been detected
Adding ESP-IDF tools to PATH...
  File "/opt/nodemcu-firmware/sdk/esp32-esp-idf/tools/idf_tools.py", line 406
    info(f'Downloading {url}')
                            ^
SyntaxError: invalid syntax
Makefile:13: recipe for target 'menuconfig' failed
make: *** [menuconfig] Error 1`

XinyuTan1985, 2023-11-06:
Same here, please help!
Entering 'components/qrcodegen/qrcodegen'
Entering 'components/u8g2/u8g2'
Entering 'components/ucg/ucg'
Entering 'sdk/esp32-esp-idf'
. /opt/nodemcu-firmware/sdk/esp32-esp-idf/export.sh && make "menuconfig"
Setting IDF_PATH to '/opt/nodemcu-firmware/sdk/esp32-esp-idf'
Detecting the Python interpreter
Checking "python" ...
Checking "python3" ...
/opt/nodemcu-firmware/sdk/esp32-esp-idf/tools/detect_python.sh: line 16: python3: command not found
Python 2.7.12
"python" has been detected
Adding ESP-IDF tools to PATH...
  File "/opt/nodemcu-firmware/sdk/esp32-esp-idf/tools/idf_tools.py", line 406
    info(f'Downloading {url}')
                            ^
SyntaxError: invalid syntax
Makefile:13: recipe for target 'menuconfig' failed
make: *** [menuconfig] Error 1

ondratu, 2024-01-03:
Problem is that SDK python tools were rewritten to python 3.x, but docker image has python 2.x. By the way, some SDK dependency is for gevent 1.5.0, which can't be compile with Python 3.11, and probably with Python 3.12 too.

jmdasnoy, 2024-01-15:
Dear @marcelstoer,

I have the same issues as documented above in using your much appreciated Docker image for building the esp32 firmware in the latest dev-esp32 branch.
The SDK python tools use the recent Python f' strings which are supported only in Python 3.6 or higher.
Your docker image provides Python 2.7.12 and can not make sense of these, hence the invalid syntax errors.
I have attempted to upgrade the Python interpreter in my running container, but even after adding the apt repository ppa:deadsnakes/ppa I can only bump Python to 3.5 which does not solve the issue.

Your help in providing an updated Docker image would be much appreciated.
Best regards,

jmdasnoy

marcelstoer, 2024-01-15:
All of the grieve is caused by #101. Please continue any discussions over there.

closed 2024-01-15 by marcelstoer